prusa3d/PrusaSlicer #11667: Slicer isn't using Extruder 2 for Infill

BrienKing, 2023-11-14:
### Description of the bug

This is a strange one as I've done this on previous prints...

I want the Perimeters to be done with Extruder 1 and the rest of the Infill done in Extruder 2.  

I have it setup this way:
[image]

As you can see, it's only using one extruder.

[image]

### Project file & How to reproduce

See attached 3MF.
[Test XL Shovel.zip](https://github.com/prusa3d/PrusaSlicer/files/13345605/Test.XL.Shovel.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 B1

### Operating system

Windows 10

### Printer model

Prusa XL Dual Extruder (alpha 2)

rtyr, 2023-11-14:
You have to set the extruder to "default" in object list.

[image]

closed 2023-11-14 by rtyr

BrienKing, 2023-11-14:
Why wouldn't setting it to 1 for that work?  I was explicitly telling the slicer which tool I wanted to use for each thing.

myParakeet, 2024-08-10:
So what is workflow if I have two objects and 5 extruders but perimeters and infill all need to be unique and not 1. How to set infill and perimeters explicitly?

Razor412, 2024-12-04:
@BrienKing and @myParakeet did you ever come up with a solution to this? I'm in a situation where I need to do a similar thing. Thanks.

rtyr, 2024-12-04:
It can be set separately per object (right click in the object list -> Add settings -> Extruders]

Razor412, 2024-12-04:
Ah, thank you! Makes sense now.