prusa3d/PrusaSlicer #11669: Please add ring geometry for support enforcer.

FrHePo, 2023-11-15:
I'm using 2.7.0-beta1.
I want to support only the upper, outer overhanging edge. Using a cylinder under this edge as support enforcer causes also the inner space to get filled with supports. I want to avoid this. Adding an inner support blocking cylinder doesn't work in combination with the above mentioned support enforcer. When I switch to "supports everywhere" also some unwanted supports get added.
I would like a ring geometry to be added for use as support enforcer. Thus enabling to define the inner ring diameter.

[image]
[SuperPINDAPrecisionAdjusterforMK3S.zip](https://github.com/prusa3d/PrusaSlicer/files/13360247/SuperPINDAPrecisionAdjusterforMK3S.zip)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2023-11-15:
Just use the support painting tools to define the faces you want to support. It’s for things like this that they were designed for and they are very effective once you know how they work. 
I haven’t had chance to open your file yet but from the looks of it you have support type set to grid, change that to type snug. That will limit the expansion of the support   

I’ll try and modify your project with an example when I get to a pc later today.

lukasmatena, 2023-11-15:
As @neophyl suggested, paint-on supports are the tool that was designed for situations like this. You can also use the Shape Gallery to add your own frequently used geometry. We cannot add a primitive geometry for each of the many use cases.

FrHePo, 2023-11-15:
Ok, got it. Thanks! Paint-on supports were new to me but it seems to be quite easy and effective to use.

FrHePo, 2023-11-15:
When I open the Shape Gallery Prusa Slicer crashes 😳
[image]

lukasmatena, 2023-11-15:
Yes, I should have mentioned that. That is a bug in 2.7.0-beta1, which will be fixed in the next release (due in several days).
Can I close the issue? It sounds to be solved to me.

FrHePo, 2023-11-15:
Yes, please close this issue. Thanks 👍

closed 2023-11-16 by lukasmatena