prusa3d/PrusaSlicer #11671: Fails to build with wxWidgets 3.2.4

nunotexbsd, 2023-11-15:
### Description of the bug

Does not build

Similar issue found on:
https://github.com/audacity/audacity/pull/5597

### Project file & How to reproduce

Full log: https://people.freebsd.org/~eduardo/logs/PrusaSlicer-2.6.1.log

```
[ 65% 339/504] /usr/bin/c++ -DBOOST_LOG_DYN_LINK -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DSLIC3R_CURRENTLY_COMPILING_GUI_MODULE -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_THREAD_SAFE -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/local/include/dbus-1.0 -I/usr/local/lib/dbus-1.0/include -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/.build/src/platform -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/hidapi/include -I/usr/local/include/gtk-3.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/cairo -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/freetype2 -I/usr/local/include/harfbuzz -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/atk-1.0 -I/usr/local/include/pixman-1 -I/usr/local/include/libpng16 -I/usr/local/include/gio-unix-2.0 -I/usr/local/include/fribidi -I/usr/local/include/libepoll-shim -I/usr/local/include/at-spi2-atk/2.0 -I/usr/local/include/at-spi-2.0 -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/.build/src/libslic3r -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/libnest2d/include -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/miniz -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/glu-libtess/include -isystem /usr/local/include/eigen3 -isystem /usr/local/lib/wx/include/gtk3-unicode-3.2 -isystem /usr/local/include/wx-3.2 -isystem /usr/local/include -isystem /wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/libigl -isystem /usr/local/include/Imath -O2 -pipe -DLIBICONV_PLUG -fstack-protector-strong -fno-strict-aliasing  -DLIBICONV_PLUG -Wall -Wno-reorder -pthread -O2 -pipe -DLIBICONV_PLUG -fstack-protector-strong -fno-strict-aliasing  -DLIBICONV_PLUG  -DNDEBUG -std=gnu++17 -fPIC -DNDEBUG -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-deprecated-declarations -Wno-misleading-indentation -pthread -DOPENVDB_ABI_VERSION_NUMBER=10 -MD -MT src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o -MF src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o.d -o src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o -c /wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/slic3r/GUI/PhysicalPrinterDialog.cpp
FAILED: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o 
/usr/bin/c++ -DBOOST_LOG_DYN_LINK -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DSLIC3R_CURRENTLY_COMPILING_GUI_MODULE -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_THREAD_SAFE -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/local/include/dbus-1.0 -I/usr/local/lib/dbus-1.0/include -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/.build/src/platform -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/hidapi/include -I/usr/local/include/gtk-3.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/cairo -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/freetype2 -I/usr/local/include/harfbuzz -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/atk-1.0 -I/usr/local/include/pixman-1 -I/usr/local/include/libpng16 -I/usr/local/include/gio-unix-2.0 -I/usr/local/include/fribidi -I/usr/local/include/libepoll-shim -I/usr/local/include/at-spi2-atk/2.0 -I/usr/local/include/at-spi-2.0 -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/.build/src/libslic3r -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/libnest2d/include -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/miniz -I/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/glu-libtess/include -isystem /usr/local/include/eigen3 -isystem /usr/local/lib/wx/include/gtk3-unicode-3.2 -isystem /usr/local/include/wx-3.2 -isystem /usr/local/include -isystem /wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/libigl -isystem /usr/local/include/Imath -O2 -pipe -DLIBICONV_PLUG -fstack-protector-strong -fno-strict-aliasing  -DLIBICONV_PLUG -Wall -Wno-reorder -pthread -O2 -pipe -DLIBICONV_PLUG -fstack-protector-strong -fno-strict-aliasing  -DLIBICONV_PLUG  -DNDEBUG -std=gnu++17 -fPIC -DNDEBUG -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-deprecated-declarations -Wno-misleading-indentation -pthread -DOPENVDB_ABI_VERSION_NUMBER=10 -MD -MT src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o -MF src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o.d -o src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PhysicalPrinterDialog.cpp.o -c /wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/slic3r/GUI/PhysicalPrinterDialog.cpp
/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/slic3r/GUI/PhysicalPrinterDialog.cpp:465:17: error: call to member function 'set_values' is ambiguous
        choice->set_values({ m_config->opt_string("printhost_port") });
        ~~~~~~~~^~~~~~~~~~
/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/slic3r/GUI/Field.hpp:381:9: note: candidate function
        void                    set_values(const std::vector<std::string> &values);
                                ^
/wrkdirs/usr/ports/cad/PrusaSlicer/work/PrusaSlicer-version_2.6.1/src/slic3r/GUI/Field.hpp:382:9: note: candidate function
        void                    set_values(const wxArrayString &values);
                                ^
1 error generated.
```

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

FreeBSD 15 CURRENT amd64

### Printer model

N/A

nunotexbsd, 2023-11-21:
I take a look at OpenBSD port and tested on FreeBSD and it builds correctly:
```
--- src/slic3r/GUI/PhysicalPrinterDialog.cpp.orig
+++ src/slic3r/GUI/PhysicalPrinterDialog.cpp
@@ -462,7 +462,7 @@ void PhysicalPrinterDialog::build_printhost_settings(C
     // Always fill in the "printhost_port" combo box from the config and select it.
     {
         Choice* choice = dynamic_cast<Choice*>(m_optgroup->get_field("printhost_port"));
-        choice->set_values({ m_config->opt_string("printhost_port") });
+        choice->set_values((const std::vector<std::string>){ m_config->opt_string("printhost_port") });
         choice->set_selection();
     }
```
and
```
--- src/slic3r/GUI/Plater.cpp.orig
+++ src/slic3r/GUI/Plater.cpp
@@ -5310,7 +5310,7 @@ void Plater::load_project(const wxString& filename)

     p->reset();

-    if (! load_files({ into_path(filename) }).empty()) {
+    if (! load_files((const std::vector<boost::filesystem::path>){ into_path(filename) }).empty()) {
         // At least one file was loaded.
         p->set_project_filename(filename);
         // Save the names of active presets and project specific config into ProjectDirtyStateManager.
```

bnavigator, 2024-08-29:
See #11769

closed 2026-09-29 by prusaslicer-bot[bot]