prusa3d/PrusaSlicer #11673: Slicing error on thin columns with 0.6mm nozzle - printing into air

NKT00, 2023-11-15:
### Description of the bug

Just swapped my nozzle from 0.4mm to 0.6mm, and found a bug.

With 0.4mm nozzle, we see a column created properly, with each layer printed on top of the next. 
[image]

But with the 0.6mm nozzle, we can see that layers are skipped, causing failure as it prints into thin air after skipping the previous layer(s).
[image]

Even if we reduce the layers to 0.15, we still see skipped layers.
[image]

The only change is from 0.4 to 0.6mm nozzle. Material is PETg.

### Project file & How to reproduce

[Column Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/13365548/Column.Bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+win64

### Operating system

Windows10

### Printer model

Prusa 3S

NKT00, 2023-11-15:
Also appears with a 0.8mm nozzle selection.
[image]

neophyl, 2023-11-15:
Its an already known issue with Arachne.  Change your type to classic and the columns appear ok.  
[image]

NKT00, 2023-11-15:
Thanks. Just weird how it only happens with the larger nozzle.

closed 2023-11-20 by Jan-Soustruznik