prusa3d/PrusaSlicer #11674: First layer height can change model geometry, not just its dimensions

pylorak, 2023-11-15:
### Description of the bug

In the attached project, the model has a flat top surface (it is also parallel to the build plate). This is also demonstrated by being able to select the top surface as a single plane prior to slicing, like in this picture:

[image]

If this model is sliced with "layer height"=0.2 and "first layer height"=0.2, then everything is fine and the model's surface is flat after slicing:

[image]

However, if the "first layer height" is changed to 0.3, then suddenly the top surface is not flat anymore:

[image]

This effectively changes the model's geometry, not just its dimensions. This is unexpected, wrong, and fatal to precision functional parts (like in this case).

### Project file & How to reproduce

Steps to reproduce:
Open attached project and slice the model.
[[a]_rear_half_g2.zip](https://github.com/prusa3d/PrusaSlicer/files/13367585/a._rear_half_g2.zip)

Results:
Top surface geometry is changed, it is not flat anymore.

Expected results:
An additional layer is either added or not, possibly changing model dimensions. But top surface curvature should remain unchanged.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64, also  2.7.0-beta1+win64

### Operating system

Windows 11 22H2 (Build 22621.2428)

### Printer model

Voron Trident

pylorak, 2023-11-15:
Additional info: Issue happens with both Arachne and Classic engines.

lukasmatena, 2023-11-15:
I think your model does not sit perfectly flat on the bed. When I use the place on face tool and apply it on the bottom, it is sliced the way you want.

pylorak, 2023-11-15:
I already tried that. If I use the place on face tool that the positions of the steps are moved but it the model is still not sliced correctly. Screenshot after placing bottom face on bed:

[image]

mix579, 2023-11-15:
When I load the STL into Fusion, it looks like the top and bottom surfaces are not absolutely parallel but off by a tiny amount, which would explain the issue you're seeing.

pylorak, 2023-11-15:
I'm also in contact with the model's creator. He said before I posted this issue that the model was already flattened with your tool (whatever he meant by that, probably imported STL to PS, used the "place on face" tool then reexported the STL from PS). I now relayed everything back to him.

neophyl, 2023-11-15:
The place to face tool does not flatten a model. 
If the top and bottom surfaces are off parallel even by a fraction which it looks like they are then ideally that should be fixed in cad. Something easily done in most cad packages. 

The only thing you can do in the slicer is use the cut function, which is not ideal as you would usually lose too much, especially when sizing is critical.

pylorak, 2023-11-15:
I am sure by "flatten" he meant "make the bottom surface sit flat on the buildplate". The non-parallelity between top and bottom came up after that. He said he will likely get to fixing the model during the weekend. I assume this ticket can be closed soon. 

AFAICT, this doesn't look like a PS issue currently. Aside from the model possibly not sitting perfectly on the buildplate after "place on face" and an export from PS (since apparently this was already done yet the model still didn't sit perfectly), but that would be a separate issue and a separate ticket.

closed 2026-09-29 by prusaslicer-bot[bot]