prusa3d/PrusaSlicer #11678: Wipe tower size is inversely proportional to layer height

dougisfunny, 2023-11-16:
### Description of the bug

I found that printing at very low layer heights causes exceptionally large wipe towers to be made. This makes colored lithophanes more problematic. To illustrate the issue, I made two Shape-Boxes, set on different extruders and show the resultant wipe towers at .1, .05 and .01 layer heights. I wouldn't expect it to be necessary to have a prime block end up twice the size of the rest of the print or more.
0.1:
[image]
0.05:
[image]
0.01:
[image]

### Project file & How to reproduce

1. Add 2+ objects
2. Set them on different extruders.
3. Set the layer height really low like .01
4. See the huge purge block in comparison to what is necessary.
[image]
[Shape-Box.Layer Height.Wipe Tower Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/13371798/Shape-Box.Layer.Height.Wipe.Tower.Issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-Beta1,2.7.0-Alpha1, 2.6.1 +win64

### Operating system

Windows 10

### Printer model

Prusa XL 5T

lukasmatena, 2023-11-16:
It has to be inversely proportional to layer height. The volume required to purge per tool-change is fixed, so the total wipe tower volume is proportional to number of toolchanges. And the smaller the layer height, the more toolchanges.

dougisfunny, 2023-11-16:
I mean, I get that, to a point. But it isn't right how it is. The ones I posted were for simplicity in recreating the issue, but I have 3 stringing tests pieces, on different tool heads, total print time for the parts: 2h20m. Print time for the purge block, 22h52m. It is supposed to be prime tower for the XL, not a purge tower isn't it?
[image]

lukasmatena, 2023-11-16:
I am confused. Are we talking about wipe tower size or the time estimate? In either case, I will need a 3MF.

dougisfunny, 2023-11-16:
Both really, I guess I just don't think the wipe tower being 85% of the print on an XL seems quite right. Maye a separate value for minimum purge, versus minimum prime?
[stringing_test_extreme wipe tower.zip](https://github.com/prusa3d/PrusaSlicer/files/13373298/stringing_test_extreme.wipe.tower.zip)

dougisfunny, 2023-11-16:
After playing with it more, I agree the title is poorly chosen. However I think there is something off in the way the volumes of the wipe tower are decided.

Changing the 'minimal purge on wipe tower' from 1 to 100 mm^3, changes the amount of filament used by a factor of 14x on an XL profile, an 5.3x on an MMU profile.

lukasmatena, 2023-11-16:
I don't see any problem. You are using extremely small layer height of 0.01 mm, which means there simply are many toolchanges.

> I just don't think the wipe tower being 85% of the print on an XL seems quite right

The layer height of 0.01 is also not quite right for the XL. But more importantly, the percentage is affected by what you print. If the object size is negligible, the wipe tower will necessarily make up most of the print and the waste/print ratio will rise, there is no way around that. It is up to the user to decide what is acceptable.

> Changing the 'minimal purge on wipe tower' from 1 to 100 mm^3, changes the amount of filament used by a factor of 14x on an XL profile, an 5.3x on an MMU profile.

First, "minimal purge on the wipe tower" is a MINIMAL purge. It can end up to be more, because the purge can not e.g. stop in the middle of the wipe tower perimeter, it has to finish it. Second, MMU profiles have ramming enabled, which means that there is extra material extruded into the wipe tower.

dougisfunny, 2023-11-17:
There is still some wipe tower oddness with very low layer heights. Why is it still so big, if it is so empty?
[image]
[stringing_test odd wipe tower.zip](https://github.com/prusa3d/PrusaSlicer/files/13385992/stringing_test.odd.wipe.tower.zip)

dougisfunny, 2023-11-17:
Figured it out. The "Multitool ramming flow" determines the first layer size, even when ramming is turned off, and even though flow is a rate, not a volume.

closed 2026-09-29 by prusaslicer-bot[bot]