prusa3d/PrusaSlicer #11684: Primetower does NOT support full contact support

TobyCWood, 2023-11-16:
Currently the prime tower does not support multiple materials which do not bond... as with PLA and PETG. We NEED to use these for full contact support. The current design of the prime tower mixes all materials and as such will always fail.

A completely separate priming tower or some method of separating the full contact support material is required.

One approach would be to have a completely separate prime tower just for the support material extruder. 

Another approach could provide a separate  column within the prime tower for this material (see how Cura is doing it!!!)

Timothy-Ecc19, 2023-11-20:
If you set the wipe tower extruder instead of allowing it to use any active extruder it will act exactly how you are saying it should. 

It would be nice if this step was not needed, but that can be solved in the future.

TobyCWood, 2023-11-21:
Don’t understand what you are referring to. We need a tower for each tool i.e. both non bonding materials. Since they can’t bond they need their own complete columns from the bed up until unneeded by the object. Please take a good look at how Cura is doing it. It’s a column with each material which has its own toolpath.Sent from my iPad

Timothy-Ecc19, 2023-11-21:
Sorry, I misread part of your comment. This does not give two seperate towers.

What I have done for printing petg with pla supports is set the wipe tower extruder to 1 instead of the default of 0 (meaning any extruder). This means extruder 1 is always used for the perimeter of the wipe tower and all pla is contained preventing wipe tower failure.

TobyCWood, 2023-11-23:
I will try thanks!

TobyCWood, 2023-11-23:
Ok I tried it. No it does not work. Doing it exactly as you describe the wipe tower includes layers which combined the PETG with the PLA. Part of the problem is the first layer raft on the tower is always one or the other with the other printed on top of it. However, there are portions of the tower where the two materials still print on top of each other. It won't work. The wipe tower has to be able to allow two materials to have completely independent tool paths from the bed to all the way up in Z.
Using the PLA/PETG combination for full contact support can and should become a standard task flow approach given the XL as a tool changer. But, Prusaslicer has to correctly support it.

Timothy-Ecc19, 2023-11-23:
I just opened up one of the files I printed which is PETG with PLA supports in PrusaSlicer-2.7.0-rc1 and I do see minimal printing of PLA/PETG on top of each other, but it has not caused me any problems. Is this picture how it looks for you?
[image]

TobyCWood, 2023-11-24:
Try it with 4 extruders of one material and the fifth with another.I get not only a raft with a different material on top, but also whole sections with the materials one on the other. Tried all the combinations of settings.Sent from my iPhone

Timothy-Ecc19, 2023-11-26:
You are right, with all five extruders I would get pla on top of petg. Definitely an issue.

Racemade, 2024-06-05:
UP
It would be an improvement of seperate wipe tower for each head/ or filament line one for petg and another one for pla.