prusa3d/PrusaSlicer #11685: Adding a new printer to an existing "physical printer" configuration causes Host Type to change to OctoPrint

brettvitaz, 2023-11-16:
### Description of the bug

Adding a new printer to an existing "physical printer" configuration causes Host Type to change to Octoprint.

### Project file & How to reproduce

1. Open "Printer Settings" tab
2. Select a physical printer using PrusaLink
3. Click Gear icon to edit physical printer configuration
4. Click "+" button to add a new printer
5. Observe "Host Type" has changed from PrusaLink to OctoPrint

This change appears to be accepted by the application and trying to upload a sliced file fails until the Host Type is changed back to PrusaLink.
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1-rc1+MacOS-x64

### Operating system

MacOS 14.1.1

### Printer model

Prusa XL

kocikdav, 2023-11-20:
Hello @brettvitaz, thank you for reporting the issue. 
The change to Octoprint here is correct, because the last printer is Creality CR-6 - a printer that does not support PrusaLink. If you change it to a printer that supports PrusaLink, the host type should change back.

On the other hand we are aware that adding first printer by alphabetical order could be misleading as well.

brettvitaz, 2023-11-20:
Thank you for the response. As a user, I do not believe this is a good solution. I do not believe that it is appropriate to change the host type automatically, especially once it is already configured, no matter what type of printer was added. This feels like too much automation and it leaves the configuration broken if the user doesn't notice the change.

brettvitaz, 2023-12-09:
A thought I just had is that it would be better to filter the list of printers to ones that are compatible with the print host instead of changing the print host to be compatible.

closed 2026-09-29 by prusaslicer-bot[bot]