prusa3d/PrusaSlicer #11686: [2.7.0-rc1] Bug in thumbnail creation

oozebot, 2023-11-16:
### Description of the bug

In the betas/release candidates for 2.7, the thumbnail image appears to be created before the UI is finished rendering when multiple objects are present. See screenshot. Rolling back to 2.6 corrects the issue.

[image]

### Project file & How to reproduce

Unfortunately, we can't provide the project files that we've experienced this with, but can say that it's happening on many of our files.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc1

### Operating system

Windows 11

### Printer model

oozeBot Elevate

lukasmatena, 2023-11-17:
Thanks. If the models are private, can you please provide a 3MF where they are replaced by cubes or something? I was not able to reproduce the problem, it may depend on your printer bed size or something.

lukasmatena, 2023-11-17:
Probably related to SPE-1868

oozebot, 2023-11-17:
OK - figured it out. Insert a model and duplicate it. Then "Set as a Separate Object" on one of the models and mirror it. That causes the screenshot to behave like we previously posted every time. Attached is a project file which demonstrates the issue.

[eastman-black-cat-v1-0.zip](https://github.com/prusa3d/PrusaSlicer/files/13391870/eastman-black-cat-v1-0.zip)

Jan-Soustruznik, 2023-11-21:
SPE-2048

lukasmatena, 2023-11-22:
Fixed in [2.7.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc2). Closing.

closed 2023-11-22 by lukasmatena