prusa3d/PrusaSlicer #11687: G-code substitutions not replacing placeholders

dmbaio, 2023-11-16:
### Description of the bug

In the 2.7 series of PrusaSlicer, machine jerk placeholders are not being properly replaced in the final g-code based on the substitution settings. Please see screenshots of my settings and the resulting g-code. This worked as expected in 2.6.

[image]

[image]

### Project file & How to reproduce

[Placeholder Issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13383830/Placeholder.Issue.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-rc1+MacOS-arm64

### Operating system

macOS Sonoma 14.2

### Printer model

AnkerMake M5

lukasmatena, 2023-11-17:
> This worked as expected in 2.6.

Has it? I have just tested in 2.6.0 and I see the same behavior as in 2.7.0-rc1. As far as I know, this was never supported, the substitutions were never processed by the placeholder parser.

dmbaio, 2023-11-17:
Well this is embarrassing. I could have sworn it did but even looking back at files sliced with 2.6 I see what you mean. I don't really know how I thought it was working correctly the whole time...however, if not a bug, could this be an enhancement request?

Abasz, 2024-01-19:
I was trying to use placeholders in the G-code substitution as well but I cant seem to make it work.

The docs says: _"With regular expression disabled, n, r, t, and placeholders are supported in both "Find" and "Replace with" patterns similarly to Notepad++'s "extended" syntax."_

But with find: `M104 S210` and replace with `M104 S{first_layer_temperature[0]}` I get:

`M104 S{first_layer_temperature[0]}` instead of the actual first layer temp (lets say 205). 

I understand that regex and placeholders are not supported but I understand from the docs that simple word replacement should work with placeholders.

closed 2026-09-29 by prusaslicer-bot[bot]