prusa3d/PrusaSlicer #11693: Ramping Lift at Tool Change

JeremiahGillis, 2023-11-17:
### Description of the bug

I am not sure if this should be a feature request or a bug. I have ramping lift enabled which I can clearly see working at a layer change. However, the standard z-axis lift occurs at the end of a layer when a tool change is the next step. This can lead to stringing. I am not using a wipe tower. Shouldn't ramping lift also occur here?

[image]

https://github.com/prusa3d/PrusaSlicer/assets/13179985/b0683d42-10b6-4fec-983a-864e16fee2df

[PrusaXL-5-Head-Toochanging-Banding-Test.zip](https://github.com/prusa3d/PrusaSlicer/files/13395140/PrusaXL-5-Head-Toochanging-Banding-Test.zip)

### Project file & How to reproduce

You can open my attached 3mf model and view the tool path with tool changes enabled.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc1

### Operating system

Windows 11 x64

### Printer model

Prusa XL, 5.1.0-alpha2

JeremiahGillis, 2023-11-17:
Here is a finished picture of the towers. You can see what I am describing above by looking at the front left of the towers, this is where the tool change happens.

[image]

rtyr, 2023-11-17:
You are right, but the z-lift you are talking about is performed by the firmware after tool change command. It is not an issue if the wipe tower and toolchange ramming is enabled (both is enabled by default), but we are planning to implement ramping z-hop to the firmware z-lift as well.

rtyr, 2023-11-17:
I am moving this to the firmware repo.