prusa3d/PrusaSlicer #11697: 2.7.0.RC1 & XL 5Toolheads - FW5.1.0Beta  - Only E2 selected, but E1 is used for priming

sternbach, 2023-11-18:
### Description of the bug

I have prepared a small cube, the standard extruder is set to E2. E1 is not used. 
When all extruders are parked, E1 is tested, heated up, purged a little and parked.
E2 is fetched, no purge, then printing starts. If there is no skirt as a kind of cleaning process, problems can occur.
E1 should never be used.

[image]
[image]

### Project file & How to reproduce

How to reproduce: 
1) XL: Park the current tool.
2) create a test file using E2 as default/selected extruder and upload/print the file. See video below:

https://github.com/prusa3d/PrusaSlicer/assets/91251599/bd42d427-28ef-439f-8e8f-90da18dab9ca

Project file:
[Shape-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13400274/Shape-Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0.RC1

### Operating system

Win11 Enterprise 23H2

### Printer model

Original Prusa XL 5 Toolheads

rtyr, 2023-11-18:
Your custom printer profile is using incorrect start g-code meant for single tool XL.

closed 2023-11-18 by rtyr

sternbach, 2023-11-18:
@rtyr Thank you for the info, but I haven't manually changed any GCode. I have used the add/remove printer function.

rtyr, 2023-11-18:
The printer profile used in your 3MF is not stock system profile, it is some user-created custom printer profile.

The name of the profile is "Original Prusa XL 5 Input Shaper (Alpha) 0.6 nozzle pws".

sternbach, 2023-11-18:
> The printer profile used in your 3MF is not stock system profile, it is some user-created custom printer profile.
> 
> The name of the profile is "Original Prusa XL 5 Input Shaper (Alpha) 0.6 nozzle pws".

Yes, this is true, the name has been changed: When adding a "physical" printer, I was asked for a name, so I added my usual suffix. That's all the magic. As an industrial sw/hw developer, I don't touch settings if there's no need for.

If I touch any lines, I'll always use the inline comments " ; #pws|description", as I do in my sources. This is/was not the case here. 
You know, "never touch a running system" and so on...

And: I'm not the only one with this problem, as seen in the prusa forum/groups...

Assumption
I have added the XL in 2.6.x some months ago, and the profiles were imported in 2.7.0 alpha1. I have added the multitool profile in 2.7.0 Alpha1, and added the physical printer for prusa link communication. I run a MK3s Bondtech and now an XL 5 TH on my developer machine, there are no other printers in my company at the moment. Maybe the problem was "imported/converted"?

Anyway, your info was very valuable, I deleted the old profiles and added the new beta profile, now everything works as expected.

Thank you and cheers,
Peter

rtyr, 2023-11-18:
My guess is, that this profile was saved as a modified copy of stock single-tool XL profile (plus manually added more extruders and renamed). The name is different from stock profile even without "pws" suffix.

sternbach, 2023-11-18:
> My guess is, that this profile was saved as a modified copy of stock single-tool XL profile (plus manually added more extruders and renamed). The name is different from stock profile even without "pws" suffix.

This could be the case. Once I was using the single profile months ago (did get my machine some weeks ago). 
But what happened then with the added 5TH alpha profile? Anyway, problem solved. Have added your info at Discord and in the forum, maybe helpful for some other users :)