prusa3d/PrusaSlicer #11703: Sliced G code for Prusa XL turns off non existent heaters causing issue in Octoprint

magiesing, 2023-11-19:
### Description of the bug

When selecting the 2 tool head version (Original Prusa XL - 2t Input Shaper) through the configuration wizard, when slicing it still adds in code to turn off tool heads, including ones that are not configured or installed (in my case, 3-5) as if every XL is a 5 tool head printer.

This is a problem when trying to print from OctoPrint, as when it executes the code (M104 T2 S0 ), it defaults to T0 and then suppresses further T0 commands, making the print fail. I have gotten around this by adding G code substitution to comment out these commands.
[image]

The commands in the G code file are on lines 1651-1653. In the screenshot you can see that I only have 2 extruders setup, using the XL 2T printer profile.

### Project file & How to reproduce

[Calibration Cube_0.6n_0.15mm_PLA,PLA_XL_47m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/13404466/Calibration.Cube_0.6n_0.15mm_PLA.PLA_XL_47m.gcode.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc1

### Operating system

Windows !0

### Printer model

Prusa XL 2T

rtyr, 2023-11-20:
It feels more like Octoprint issue (it imho should just ignore those commands as the firmware does), but we will add additional condition to not emit these commands if there is no more than 2 tools.

rtyr, 2023-11-21:
Start g-code was updated in configuration update 1.11.4. Closing.

closed 2023-11-21 by rtyr