prusa3d/PrusaSlicer #11704: Access violation

Fuzeboi, 2023-11-19:
### Description of the bug

[image]

I wanted to print the master sword from printables "Legend of Zelda Master Sword (BOTW) (TOTK) (Full Scale)", loaded all the blades into prusaslicer and chose ultradetail, 0,07 mm and it crashed. the same thing happened for 0,05 mm.

the image is what sometimes showed up before prusa crashed

I have a Razer 15 from 2020 which works fine so I don't think it's my performance on my computer that lacks.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1 based on Slic3r

### Operating system

windows 11

### Printer model

prusa mini +

Fuzeboi, 2023-11-19:
[image]

Here is a picture of prusaslicer with parts and all

Fuzeboi, 2023-11-19:
here is the parts: https://www.printables.com/model/252630-legend-of-zelda-master-sword-botw-totk-full-scale/files

rtyr, 2023-11-20:
Could you please provide your 3MF project file?

Fuzeboi, 2023-11-20:
[Blade Piece 6.zip](https://github.com/prusa3d/PrusaSlicer/files/13419245/Blade.Piece.6.zip)

rtyr, 2023-11-20:
Thanks. I can reproduce crash with PS2.6.1, but it slices fine in 2.7.0-rc2.

Fuzeboi, 2023-11-20:
I tried it too and it works fine in the pre-release. It seems to be a problem with specifically Blade piece 6, without that one it worked fine but with just that one it crashed

lukasmatena, 2023-11-21:
@Fuzeboi Thanks. In that case, it is probably a duplicate of something we already fixed, possibly #11267. It is probably not worth to investigate further if nobody can reproduce it in 2.7.0-rc1. Closing.

closed 2023-11-21 by lukasmatena