prusa3d/PrusaSlicer #11708: PrusaSlicer 2.7.0-rc1 Prusa XL wipe tower brim is deformed

Timothy-Ecc19, 2023-11-20:
### Description of the bug

There is a strange movement during the creation of one of the brim lines for the wipe tower when using soluble full.

[image]

### Project file & How to reproduce

[Strange wipe tower behavior.zip](https://github.com/prusa3d/PrusaSlicer/files/13406166/Strange.wipe.tower.behavior.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.7.0-rc1

### Operating system

Windows 10

### Printer model

Prusa XL five toolhead

Jan-Soustruznik, 2023-11-21:
SPE-2049

lukasmatena, 2023-11-22:
Thanks. Fixed in [2.7.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc2). Closing.

closed 2023-11-22 by lukasmatena