prusa3d/PrusaSlicer #11709: PrusaSlicer 2.7.0-rc1 "Use Ramping Lift" should force "Retract On Layer Change" to On

dougemon, 2023-11-20:
### Description of the bug

I just ran my first print on a MK3S sliced in prusaslicer 2.7 and wanted to take advantage of the new ramping with helical screw movement.
In this profile I had beforehand, I checked the box enabling Use Ramping Lift.
What I noticed on the print, is there were these zits on the exterior. It turned out they line up 1 for 1 with where the layer change ramp with the new helical movement happened.
What I needed to do to fix this was to check the box that says "Retract on Layer Change"
My feedback is this should be locked as checked if the user allows Ramping Lift.
[image]
[image]

### Project file & How to reproduce

[soap dish helical screw defects.zip](https://github.com/prusa3d/PrusaSlicer/files/13406503/soap.dish.helical.screw.defects.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc1

### Operating system

Windows 10

### Printer model

Prusa MK3S+

neophyl, 2023-11-20:
Got to agree.  I normally don't retract on layer change either, tried the ramp option and got little loops of filament sticking out of the side of the print, not just little bulges.  Turned retract on layer change on and reprinted and the loops were gone.  

Then again a tiny little gap appeared in the perimeter at the layer change, so I went back to my original profile with no retract on layer change and old style z hop and the perimeters were once again perfect.  Which is why I originally had retract on layer change turned off in the first place.

lukasmatena, 2023-12-14:
Fixed in [PrusaSlicer 2.7.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1). Closing.

closed 2023-12-14 by lukasmatena

johngalt01, 2023-12-19:
this is not fixed. these helical ramp things need an option to completely disable.

neophyl, 2023-12-19:
[image]
Printer Settings>Extruder Untick the Use ramping lift.  Its been possible to disable them since they were added.

There are also settings on the filament profiles Overrides page than can be used too
[image]

neophyl, 2023-12-19:
Well it looks like for everyone else that when unticked it stops the helical lift.  Its possible you may have a corrupted config where the setting is stuck 'on' no matter what the gui is showing.  Although how likely that is I couldn't say. I have seen similar things in the past on older versions.

You could save a project file from PS, zip up the 3mf produced and attach it here.  That way someone else could load it in on their machine and see if it slices with the helical travel for them as a 3mf saved from PS should contain a copy of all the settings used.

johngalt01, 2023-12-19:
i guess i will uninstall 2.7.1 from my other machine and clean out older versions and try a fresh install.
i have a separate computer with 2.6.1 loaded on it for now just to get this project done.

now if i turn off ramping does it mess up other settings of retraction or does it really revert to a 2.6.X style of Zhop?

johngalt01, 2023-12-20:
thank you, reinstalling fixed the issue. now my ramping will turn off.