prusa3d/PrusaSlicer #11712: Moving object with arrow keys is not working

SCiunczyk, 2023-11-20:
### Description of the bug

Moving an object using the arrow keys no longer works in version 2.7.0-rc1

### Project file & How to reproduce

Select the object and try to move it with the arrow keys. In version 2.6 it worked.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.7.0-rc1

### Operating system

Windows 11

### Printer model

SecKit 2

rtyr, 2023-11-20:
Could you please share a screenshot of the whole PrusaSlicer window at the moment of non-working moves?

SCiunczyk, 2023-11-20:
Yes sure:

[image]

Jan-Soustruznik, 2023-11-21:
Hello, @SCiunczyk , 

When you re-open your 3MF file in latest PS version, is problem still occur?  If yes, share your 3mf file, please? 

Thank you, have a nice day

SCiunczyk, 2023-11-21:
Hi, @Jan-Soustruznik ,

when I reopen the project arrow keys are working as expected. Honestly I can't replicate the problem any more. Maybe was something with my PC? We can close the issue I think.

Have a nice day too. And thanks for your effort.