prusa3d/PrusaSlicer #11713: Over-ride perimeter passes for seclected features, walls or holes on the part.

VH-fixer, 2023-11-20:
**Is your feature request related to a problem? Please describe.**
When printing parts we often need one wall or the holes stronger.  To do this we increase the number of perimeter passes.  To reinforce 1 wall, all profiles will get the number of perimeters set.   
In many casses we just want extra perimeters on adjoining walls or around holes where bolts or inserts will be used.  

Unneeded perimeters use filament and add time. 

**Describe the solution you'd like**
In the slicer we need to ability to assign perimeters to individual walls or surfaces.  This would over-ride the default perimeter setting for the selected wall.   Example, on a part be able to use the default 2 perimeters, but on the holes in the part use 5 perimeters. 

**Describe how it would work**
There are many ways this could be implemented.
A. Have a seperate Perimeter setting for external and internal walls.  Holes have an internal wall.  A slot or pocket would be internal walls.   This would not be perfect but may cover 50% of the need.  Currently there are seperate settings for top and bottom layers.  There needs to be internal/external profile pass settings. 

B. Be able to use something similar to painting supports,  Pick a continuous flat wall, cylinder or curve surface and set profile passes that over-ride the default.   This would give the most control. 

C. Enable a setting on the modifier feature to NOT generate profiles for the modifier shape.  When Perimeters is the only change or modifier used.  If I use a  rectangle modifier currently, the area inside the rectangle has all the holes modified with the number of profiles set in the modifier, this is good.  The Bad part is the profile of the modifier block also generates profiles.  I understand this may be needed if infil is changed, but when the infill settings are deleted and only perimeters are set, the perimeters of the modifier box are not needed. 

**Describe alternatives you've considered**
Adding modifier geometry.   This tends to be cumbersom and slow.  If I try to cover multiple holes then I end up with extra profiles in the part.  If I adapt the geometry to fit the hole, I have to copy and locate it to every single hole.    They also add unwanted profiles within the part.   Using a modifier to do exterior walls will take many modifier shapes and still may have unwanted profiles.

**Additional context**

Tupson444, 2026-09-18:
still relevant

VH-fixer, 2026-09-19:
Still relevant for better control and the ability to optimize the path