prusa3d/PrusaSlicer #11718: PrusaSlicer 2.6.1 does not respect Tool Change, Emit Temperature commands automatically with Single-Extruder MultiMaterial

codefaux, 2023-11-21:
### Description of the bug

As in the title.
Windows, PrusaSlicer 2.6.1

I'm working through setting up Single-Extruder Multi Material in a spare profile, and in sanity checking the compiled GCode, I'm finding M104 commands at filament change despite having "Emit Temperature commands automatically" unchecked.

During writing up the issue, I noticed it also forces T0/T1/T2 style toolchange macros even when those are changed in Custom Gcode with Single-Ext MM setups.

### Project file & How to reproduce

I'd rather not share the project as it's a LOT more than is required. I suspect this was an overlooked gcode emission during the changes made to support those of use who disable "Emit Temperature commands automatically"

But, how to reproduce;
- Disable "Emit Temperature commands automatically"
- [image]

- Enable Single-Extruder Multi Material
- [image]

- No Toolchange gcode since I'm doing this manually
- [image]

- Slice anything with a filament change
- [image]

- Examine gcode
- [image]

For toolchange code, definitely seems like you need either a "Do not emit toolchanges automatically" option under the "Single-Extruder MM Setup" window or to modify gcode emission to respect Custom G-code.

For temperature code, seems like it was missed during "Do not emit temperature commands automatically" work.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 10 22H2

### Printer model

Ground-up custom, Klipper

lukasmatena, 2023-12-06:
The 'Emit Temperature commands automatically' option is only connected to the Start G-code. When unchecked, you basically say "My custom start G-code will take care of heating the extruders, don't do it for me at the beginning". You cannot use it to suppress temperature changing commands after toolchanges, those will be emitted unconditionally. This is not a bug.

codefaux, 2023-12-06:
Well, that covers half the bug I reported. I feel it's strange to allow us to disable temperature commands *only* in start code, who needs to disable it there for a sane reason but not elsewhere? Guess I'll go back to replacing temperature commands with a stub macro firmware-side.

Anyone care to address the other half of the bug report? The one about toolchange commands?

closed 2026-09-29 by prusaslicer-bot[bot]