prusa3d/PrusaSlicer #11719: Slicer 2.7.0 rc-1  upload of old project to MK4 5.1.0 beta : "Not a Gcode"

beng27, 2023-11-21:
### Description of the bug

First try of 2.7.0 rc1. Stock MK4 running 5.1.0 beta , is working well with previous version of slicer.  

Opened & Resliced a project created last year. Tried to upload a model from it (by wifi) and get message
"Error uploading to print host:
HTTP 415: 415: Unsupported Media Type
Not a GCODE"

But if I copy & paste the model from that project into a new project, then it uploads OK. Same printer settings in both projects.

### Project file & How to reproduce

[image]
This file is the model copied & pasted into a new project & works
[elevator. No2 L39.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13429649/elevator.No2.L39.3mf.zip)

This file is the same model in an old project - yes it has been resliced - but doesnt work.
[Full Set February 2022.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13429675/Full.Set.February.2022.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc1-McOS-x64

### Operating system

OSX 12.7.1

### Printer model

MK4, stock.

kocikdav, 2023-11-22:
Hello @beng27, from your screenshot it seems to me, that you are uploading file named "Files for printing scoops" without any extension. It should be present with default name of the upload.
[image]

BrienKing, 2024-02-14:
This is most likely due to a long filename.  I have this issue on the XL as well and when I shorten it down it works fine.

For example:
_3DBenchy_-_Dualprint_-_Gunwale__Deck__Plate__Wheel__Frames__Roof__Chimney_top_-__3DBenchy_0.4n_0.25mm_PLA,PLA,PLA,PLA,PLA_XLIS_1h38m.bgcode

Is way too long and the printer is not happy about it.  It returns the same error you get.

If I shorten it down to:
3DBenchy_0.4n_0.25mm_PLA,PLA,PLA,PLA,PLA_XLIS_1h38m.bgcode

Then it's happy.

ex-nerd, 2024-02-18:
> This is most likely due to a long filename

I just ran into the long filename thing. That "not a gcode error" is … not helpful. Exported the gcode just fine and then uploaded the file through Prusa Connect and it gave me an appropriate "name too long" error.

reinux, 2024-06-01:
It would be nice to have a better error message for this. Just had the same issue and got stuck until I landed on my message from six months ago.

ex-nerd, 2026-09-18:
Keep this open. It can't be tested on v3 until the app has been updated to support Prusa Link (or whatever will replace it in upcoming firmware).