prusa3d/PrusaSlicer #11720: PrusaSlicer-2.7.0-rc2+linux-x64-GTK3: `Gtk-CRITICAL` errors

RichiH, 2023-11-22:
### Description of the bug

When starting up PrusaSlicer-2.7.0-rc2+linux-x64-GTK3-202311211232.AppImage (and other images and binaries), there is always a flood of `Gtk-CRITICAL` & `GLib-GObject-CRITICAL`. Simply starting the slicer and not doing anything, this is the output:

````
(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.602: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.602: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.603: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.603: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.606: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.606: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.607: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.607: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.610: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.610: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.611: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.611: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.890: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.890: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.905: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.905: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.905: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.905: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.907: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.907: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.908: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.908: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.909: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.909: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.909: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:9569): GLib-GObject-CRITICAL **: 10:26:50.909: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:9569): Gtk-CRITICAL **: 10:26:50.909: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

````

I had always assumed this was known and ignored, but looking through the issues, it seems it has not been reported once. If the slicer is not started from the command line, it might be wholly invisible to most everyone so I decided to report it.

### Project file & How to reproduce

Just start Prusa-Slicer on Linux.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0-rc2

### Operating system

Debian unstable

### Printer model

XL

RichiH, 2023-11-22:
I clicked the two tasks as 'done' in case there's a filter on when people start looking at it. In this specific case, neither is actually needed.

lukasmatena, 2023-11-22:
Actually, this is a duplicate of #4935. You are basically right about the current status being "known and ignored". Closing as a duplicate.

closed 2023-11-22 by lukasmatena