prusa3d/PrusaSlicer #11723: Crashing on Fedora 38

laforestlabs, 2023-11-22:
### Description of the bug

Random crashs, sometimes while slicing, othertimes while running in the background. ocassionally the program lasts long enough to slice and export, but sometimes it will crash too quickly to generate gcode.

this has been an issue in every build of 2.7

### Project file & How to reproduce

open prusaslicer
load file
wait for crash
[Cetus Z axis endstop - mount V0.2.zip](https://github.com/prusa3d/PrusaSlicer/files/13442717/Cetus.Z.axis.endstop.-.mount.V0.2.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0-rc2+linux-x64-GTK3-202311211232.AppImage

### Operating system

Fedora 38

### Printer model

Klipper

Legion495, 2023-11-25:
This might be a potential workaround in the meantime. Something I would try at least.

Have you tried the Flatpak version too? Chances are this will work.
It is in the Flathub repository so it should be available through your Software Manager.
I am on Fedora 39 now and use the Flatpak instead of the Appimage. I have yet to encounter a issue.

laforestlabs, 2023-11-28:
Thanks for the support. I tried using the release flatpak version and haven't had any crashes yet. 

Thanks!

laforestlabs, 2024-11-21:
Thanks for the reminder, I've had no issues with the flatpak. Closing this ticket

closed 2024-11-21 by laforestlabs