prusa3d/PrusaSlicer #11726: Bed Origin for Custom Bed Shape STL

LostFool, 2023-11-22:
With the rectangular option we can have origin at 0,0 in a corner. But with a custom STL I can not change the bed origin and the help document requires it is in the center. This causes the printer to attempt to go to a negative position which does not work! 

Either the option to move the origin or accepting a bed with origin at bottom left corner like most printers would be the solution.

lukasmatena, 2023-11-23:
The origin of the STL is used as the origin of the bed. You should translate the model accordingly before you save the STL.

LostFool, 2023-11-23:
I have placed the origin at the front left corner, tried both the top and bottom corner. I always get an error, see attached. But if I follow [the help document](make sure you have the ercf compatible macros) and place the origin at top-center. It works.
[image]

lukasmatena, 2023-11-23:
Can you share the stl that is giving you the assert?

closed 2026-09-29 by prusaslicer-bot[bot]