prusa3d/PrusaSlicer #11728: Bad allocation memory crash when using spiral vase mode

lyngesdk, 2023-11-23:
### Description of the bug

[image]

I get this error when slicing the attached 3mf file

### Project file & How to reproduce

Problem occurs when I press the slice button in the attached project:
[tree.zip](https://github.com/prusa3d/PrusaSlicer/files/13446696/tree.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-rc2

### Operating system

Windows 10 business 64bit, build 19045.3693 

### Printer model

Prusa Mini+ with 5.1.0 firmware

MichaelJLew, 2023-11-23:
No problem slicing it on M1 Mac using PS 2.7.0 beta 1.

lukasmatena, 2023-11-23:
Thanks, reproduced. We will look into it.
SPE-2052

lukasmatena, 2023-11-23:
It should be fixed in [2.7.0](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0). Closing.

closed 2023-11-23 by lukasmatena