prusa3d/PrusaSlicer #11729: object based blocking of supports/ block support everywhere on object

07pepa, 2023-11-23:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when i am doing bulk printing in one color of multiple object and one needs supports (https://www.thingiverse.com/thing:2941948) 
and other like this one https://www.printables.com/cs/model/257845-print-in-place-iris-box-locking does not need it or is broken by it.

painting it is viable but for some models that has cavity where i can't see without slicing it and scrolling throught layers is risky and unreliable since you are doing it manualy checking it layer by layer... aka can take time and great deal of focusing

**Describe the solution you'd like**

Object (model) based disabling of supports

**Describe how it would work**

in GUI i mark object(model) as block all support or do not use support. That will mark it internaly with boolean and  before  its processed for  support boolean is checked and support generation is potentialy skiped if bolean is set...

it would be mutualy exclusive with paint on supports*... 

**Describe alternatives you've considered**

painting on suppots to block it  may miss a spot i can't see... unless block support  everywhere button is added that would pain everything. I would accept such solution but as programer i can se other one as more performant since whole object even does not have to be proccesed for support 

*it may be mix of two... if user does not touch it after painting blocking everywhere use boolean route otherwise convert it to everything is painted and user can over paint it with enforced support... that is best of both words. but i suggested only non mixed approach as more simple but its on you how to implement it

**Additional context**
this happens for some specific models where there is small cavity and where support may be generated but not obvious at glance

lukasmatena, 2023-11-23:
> in GUI i mark object(model) as block all support or do not use support

That has been implemented for years. Right-click the model in the right panel, select `Support material` and you can customize basic support settings per object. Even more object-based settings are available through the `Add settings` item.

Does it solve the issue?

07pepa, 2023-11-23:
yes it does and thx  and sorry for spam .... using prusa slicer for years still learning something new

closed 2023-11-23 by 07pepa