prusa3d/PrusaSlicer #11733: Multiple Extruders - Allow Selection Of Extruder For Support Material With User Selected Z-Gap Plus Wipe Tower Plus Purging Volumes

inventabuild, 2023-11-23:
Sometimes dissimilar materials like PLA for FLEX with a small z-gap are better than dissolvable supports because dissolvable supports are moisture sensitive, messy to removeSometimes dissimilar materials like PLA for FLEX with a small z-gap are better than dissolvable supports because dissolvable supports are moisture sensitive, messy to remove and take time to work with whereas PLA supports can be removed cleanly immediately from a FLEX part.  In order to do this, for example, I need to select FLEX for EXT 1 and PLA for EXT 2, but when I do that I lose the wipe tower unless z-gap is 0 and I don't want to set z-gap to 0 to get the Wipe Tower in my example.  Instead I want to set z-gap to .1, but when I do that I lose the Wipe tower.

In addition, sometimes I want to increase the Purge volumes switching between EXT 1 and EXT 2 when EXT 1 is the part and EXT 2 is the infill.

Therefore, please allow the selection of the extruder for support material while still keeping the wipe tower for any z-gap and allow purging volumes for infill start and stop. and take time to work with whereas PLA supports can be removed cleanly immediately from a FLEX part.  In order to do this, for example, I need to select FLEX for EXT 1 and PLA for EXT 2, but when I do that I lose the wipe tower unless z-gap is 0 and I don't want to set z-gap to 0 to get the Wipe Tower in my example.  Instead I want to set z-gap to .1, but when I do that I lose the Wipe tower.

In addition, sometimes I want to increase the Purge volumes switching between EXT 1 and EXT 2 when EXT 1 is the part and EXT 2 is the infill.

Therefore, please allow the selection of the extruder for support material while still keeping the wipe tower for any z-gap and allow purging volumes for infill start and stop.

Jan-Soustruznik, 2023-11-24:
Hello, @inventabuild, 

I hope that I correct understood, but I think that this setting already exist in the slicer _Print Settings -> Multiple Extruders -> section Extruders on top_  - there can be chose extruder for print support material without loose the Wipe tower.

inventabuild, 2023-11-25:
In the Wipe tower section if you have Wipe tower Enabled and you go to the Extruders section above it and you try to change the Support material/raft/skirt extruder from 0 to 1 or 2 or whatever you will get a popup that says:

"The Wipe Tower currently supports the non-soluble supports only if they are
printed with the current extruder without triggering a tool change. (both
support_material_extruder and support_material_interface_extruder need to be
set to 0).
Shall I adjust those settings in order to enable the Wipe Tower?
                                                                                                   Yes   No"
                  
  If you select Yes the extruder number gets set back to 0.  If you select No the Wipe tower is Un-enabled.  So you cannot select an extruder number (other than 0) for Support and use the Wipe tower at the same time unless you use Soluble support which means the Z-gap needs to be set to 0.  This should be changed to allow you to select whatever extruder number you want to create Support w/ a Z-gap > 0 while still printing the Wipe tower.

LukeAllen, 2025-01-07:
The error happens for me too:
On a multi-extruder Prusa XL, I'm trying to print a TPU part, using PLA for supports (because the two materials separate cleanly). But it refuses to generate a wipe tower if I explicitly set the support material. I don't see why it can't wipe on material change, regardless of the reason for the change.

MichaelEhrlichman, 2025-01-23:
I like to use PVA breakaway supports for PLA.  The supports work best with a small top contact z-distance of 0.05.  Unfortunately, I cannot have a support extruder, Top Contact Z-distance of 0.05, and Wipe Tower all enabled at the same time due to the popup referenced above.  This limitation is significant.

LukeAllen, 2026-09-21:
This isn't fixed as of the latest normal release. The Flathub install instructions for the Linux alpha release fail with a dependency error so I haven't tested the alpha (btw I really don't like Flathub as a distribution method; it is not standard).