prusa3d/PrusaSlicer #11734: Disable BGcode by default

dirkvranckaert, 2023-11-23:
I just upgraded to PrusaSlicer 2.7.0 and I'm getting the binary gcode feature now.
However that feature is not usable for me since I'm (like many) an OctoPrint user.

That being said I now have to turn the feature off every single time I use PrusaSlicer, which edits my print settings. I either have to do this all the time OR I have to create a copy for all the print settings...

I would rather have a program-wide setting to disable/enable the usage of binary gcode all the time.

mashbrno, 2023-11-24:
Neither Prusa MK4 support them on stable firmware. Unfortunatelly I haven't found even the option to choose the G-code format, so I am completely stuck. Where did you find that settings?

rtyr, 2023-11-24:
> Neither Prusa MK4 support them on stable firmware.

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

dirkvranckaert, 2023-11-24:
> Where did you find that settings?

You can find it in the _Print Settings_ tab under section _Output options_
[image]

Jan-Soustruznik, 2023-11-24:
Hello, @dirkvranckaert , 

You can create own preset with binary gcode feature disabled.

I think that we can close this issue or do you have some additional questions? 

Thank you, have a nice day.

dirkvranckaert, 2023-11-24:
@Jan-Soustruznik That's the entire point, having the option within the preset makes that I have to create my own presets of all of them, just because I want the output format to always be `gcode` instead of `bgcode`...

I think it's safe to say that either you use the good old `gcode` which is perfectly fine (although PrusaConnect suffers with it) or that you fully switch to `bgcode`. But it pointless in having to clone all of the _print settings_ presets just to toggle that flag which is not a print setting flag (it does not change anything about the print, only to the file generation).

It would make more sense for that option to be either in your general PrusaSlicer settings OR in the printer profile settings (I can imagine if you have different printers/setups that sometimes you want it, sometimes you don't but that it will more depend on the printer you are slicing for).

neophyl, 2023-11-24:
Have to agree, it makes much more sense for it to be in the Printer settings or even physical printers.

dirkvranckaert, 2023-11-24:
It's currently a pain in the ass for me, even thinking about downgrading my PrusaSlicer gain until there's a _fix_...

ericvenneker, 2023-11-24:
This option caught me off-guard after a prusaslicer upgrade. While I understand the reasoning to put it in the profiles (print settings or printer settings) it would have been nice to include an application level setting to globally force bgcode or disable bgcode.

neophyl, 2023-11-24:
A global setting wouldn't work because of the use case Dirk mentioned, namely that people can have multiple printers, some of them might support bgcode and others not.  So you would have to define it per printer.

ericvenneker, 2023-11-24:
A print(er) se.tting can co-exist with a global setting and a global setting allows a user to decide if this option should always be on or off. And if you require more fine-grained control, you can use the print(er) settings.

In my relatively basic use case (one printer, 3 nozzle diameters) it would mean 18 custom print settings profiles just to permanently change one setting. And i am ignoring the extra print settings profiles you get when have one or more IS printer profiles. That seems a bit much, but that maybe just be me 😉.

dirkvranckaert, 2023-11-25:
@ericvenneker that is why putting it in the printer settings you would turn it off for every printer, in your case that would be only 2 times (for regular and for IS printer profile). That would much better.
But indeed the current solution does not work…

rtyr, 2023-11-25:
It´s likely already possible to use bgcode with Octoprint. https://github.com/OctoPrint/OctoPrint/issues/4900

ericvenneker, 2023-11-25:
@dirkvranckaert I have 3 nozzles, so it would be 6 profiles, and that is for my situation. If you have all current nozzles (6) it would be 12 profiles. And what if you have multiple printers (2 or perhaps 3), it adds up pretty quickly.

I do agree that the setting should be at least moved to the printer settings profile. But I also believe a global setting should exist for the binary gcode option. (i.e. use _printer settings, always on, always off_)

ps-igel, 2023-11-25:
To add my two cents. @Jan-Soustruznik Might this actually be a bug? Because the tooltip clearly says it is false by default when clearly it is turned on by default.

[image]

rtyr, 2023-11-25:
> Might this actually be a bug?

@ps-igel no, you are refering to application default, not default for Prusa printers

timur-tabi, 2023-11-25:
Enabling bgcode by default in any profile is a mistake.  The feature only works when the print is on an SD card, and no one complains about running out of space on an SD card.  In addition, it requires a firmware upgrade and any tools that look for .gcode will be confused.  There is also no improvement in print quality when using bgcode.  All you're doing is causing confusion for people who use OctoPrint.

All of the profiles should be updated to turn this feature off by default.

fly74, 2023-11-26:
I agree with @timur-tabi. It took me some time to find the option to disable bgcode for a print. But the thing is, you have to do it every time. It is really annoying. Please give us an option to disable it globally. Maybe in the future octoprint will support binary gcode but in the moment it is not.

anghelos, 2023-11-27:
I agree that this should be a Printer Setting, and not a Print Preset.

robwaat, 2023-11-28:
This shouldn't have been rolled out as default until it is at least supported by the stable firmware release. It should then be an "opt-in" global setting for those using their printers over wifi.

Please rectify in v2.7.1

Shinomorisama, 2023-12-01:
I'm in the same boat on this one.  I very much dislike the binary gcode default.   I applaud the effort to solve problems, but this roll out is horrible.   It should definitely be under the printer settings and default to off.

furyus, 2023-12-03:
Another frustrated MK4 and Mini+ owner here. I love love love Octoprint and was so happy to see Prusa finally implement support in the new firmware, but BGCode defaulting to ON is so annoying! We just need a global toggle that remembers between sessions please (so that we can change it to BGCode OFF.

Shinomorisama, 2023-12-04:
Alternative option: Have bcode in printer option with default off.  With the addition that when you are sending a file over wire to physical printer it could have a checkbox to convert to bgcode since that is really the main place it is of any benefit.

fly74, 2023-12-04:
> Alternative option: Have bcode in printer option with default off.  With the addition that when you are sending a file over wire to physical printer it could have a checkbox to convert to bgcode since that is really the main place it is of any benefit.

I think that is not possible. After slicing you have gcode or bgcode generated. After that you can't choose the type. It must be an option when slicing starts.

Shinomorisama, 2023-12-04:
> I think that is not possible. After slicing you have gcode or bgcode generated. After that you can't choose the type. It must be an option when slicing starts.

Prusaslicer 2.7 has a built in converter for converting between gcode and bcode both directions. Under File>convert.   It could use that same tool to do the conversion before upload since the network transfer is the part where file size is of concern.

timur-tabi, 2023-12-08:
Even the developer of the bgcode plugin for Octoprint thinks defaulting to bgcode is a bad idea:

> My personal opinion is that the best solution is to disable the new format altogether in your slicer profiles, as seen in the screenshot below.  There is no good reason why anyone would want to use this new format with OctoPrint. The improvement in upload time would be countered by the extraction time of the ascii gcode upon upload.

https://plugins.octoprint.org/plugins/bgcode/

Frankly, we are long overdue for a response from Prusa on this issue.

ChrisHeerschap, 2023-12-08:
Have to agree that turning this on by default is a big misstep. Worse yet, the error that I got referred to alpha firmware, something I'm not running on my important printers. Only later did I realize that 5.1 had been released, but even at that, I don't see why this is being mandated as on. 

Sure, I can create my own profiles, but each time there are new ones I'll now need to compare my modified ones to the updated ones to make sure that the only thing that's changed is turning off bcode? I use stock profiles for a reason and don't want someone else's bad mandatory decision to be a reason for me to stop.

From what I'm gathering, binary gcode is a fix for slow wifi transfers and as such is a really niche need - making an edge case the default? Please no.

furyus, 2023-12-08:
@mikolaszuza Will you please help us out with this one? 

I think having BGcode ON by default is a symptom of Prusa's reluctance to support Octoprint and hard-headed pushing of PrusaConnect. When we were all frustrated by not having Octoprint support for the MK4, Prusa support told us "we don't make Octoprint - you should try PrusaConnect instead!" and that felt very tone deaf, especially when we learned how terribly slow PrusaConnect uploads were, and that it was missing critical features in comparison to Octoprint and its plug-in ecosystem.

Now we finally have Octoprint support (thank you!!!), but we have this annoying BGcode default to remind us that Prusa would rather we used their product ("which is faster now with BGcode!"). Please, please listen to your users. Thank you!

edwardhorsford, 2023-12-09:
Very frustrated here. Should I really need to edit 24+ profiles to disable a feature on my printer? And then have those show as duplicates in addition to the default profiles?

If this setting was on the printer settings it would make more sense. But given PrusaSlicer already knows people are using Octoprint, this feels like quite a punitive change.

mikolaszuza, 2023-12-10:
AFAIK, in the next build, it will be in the Printer settings.

ripienaar, 2023-12-13:
Really bad default setting, and the dialog box warning you its saving bgcode doesnt let you cancel so you can undo this silly setting without first putting down files you dont want.

furyus, 2023-12-13:
@mikolaszuza Thank you very much! We appreciate your help.

mikolaszuza, 2023-12-14:
There will also be a new global switch in Preferences->Other, which controls whether binary G-code will be generated for printers which support it. So it will be much easier to turn the feature on or off without doing any changes in profiles.

edwardhorsford, 2023-12-14:
> There will also be a new global switch in Preferences->Other, which controls whether binary G-code will be generated for printers which support it. So it will be much easier to turn the feature on or off without doing any changes in profiles.

Thanks a lot @mikolaszuza! That sounds ideal.

mikolaszuza, 2023-12-14:
Released ⤵
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1
@edwardhorsford The thanks goes to the PrusaSlicer team, they listened to your feedback here and implemented the change as fast as possible. I am just the messanger of the news. :)

lukasmatena, 2023-12-14:
Thanks for everyone's feedback. As @mikolaszuza has already written, the behaviour was changed in [PrusaSlicer 2.7.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1), hopefully for the better.

Closing.

closed 2023-12-14 by lukasmatena

dirkvranckaert, 2023-12-14:
Big thanks for the entire PrusaSlicer team here! Much appreciated! \cc @mikolaszuza @lukasmatena

fly74, 2023-12-14:
Thx a lot!

jneilliii, 2023-12-14:
I did make a plugin for OctoPrint to extract standard gcode from this format, so it is supported technically in OctoPrint now.

ChrisHeerschap, 2023-12-16:
Wait hang on, I thought this was fixed in 2.7.1. Looking at the "Original Prusa MK4 Input Shaper 0.4 nozzle" default profile, it's very clearly still enabled:

[image]

At least now I don't have to create a bunch of different filament profiles, and it's easy enough to create a new printer profile that's only modifying that, but this didn't technically get fixed, did it?

andrewjmetzger, 2023-12-16:
Hey @ChrisHeerschap :)

In addition to the printer-specific toggle for those that want its granularity, PrusaSlicer v2.7.1 _also_ adds a toggle in the global (application) preferences: 

To change the global setting, head to: Configuration > Preferences (or <kbd>CTRL</kbd> + <kbd>P</kbd>) > Other > Uncheck "Use binary G-Code when the printer supports it". Screenshot:

[image]

Hope this helps you!
Andrew

ChrisHeerschap, 2023-12-16:
@andrewjmetzger it does, thank you!