prusa3d/PrusaSlicer #11735: Prusa Slicer 2.6.0 and 2.7.0 crashing with VMWare Workstation 17.5

Bitu69, 2023-11-23:
### Description of the bug

Crashing the VMWare host completely when launching the slicer software.

### Project file & How to reproduce

Software uninstall and reinstall doesn't help. Each time a crash after launch.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1 or 2.7.0

### Operating system

Windows 10 64 bit

### Printer model

XL

timur-tabi, 2023-11-24:
I'm running 17.5 with a Linux host and Windows 10 guest.  I just installed PrusaSlicer 2.7 and it works fine.

lifeisafractal, 2024-01-16:
I have this issue too. I often get "ISBRendererComm: Lost Connnection to mksSandbox" when starting PrusaSlicer. Not every time, but often enough that it's really annoying. I'm on Win11 host and Ubuntu 23.10 guest. FWIW, I think this is more an issue with VMWare, but I'm curious why PrusaSlicer induces it so often.

Stephan-Cl, 2024-11-12:
Got that same issue as well. Latest VMWare workstation (17.5.0).
[image]

timur-tabi, 2024-11-12:
The latest is 17.6.1: https://softwareupdate.vmware.com/cds/vmw-desktop/player/

Stephan-Cl, 2024-11-12:
@timur-tabi 
Well... thanks for the tip! I had downloaded 17.6.1 but it wasn't yet installed. 
Just installed it.
Surprisingly, the issue seems to be resolved.

orionar1, 2024-11-25:
VMware Workstation 17.6.1 does not appear to fix the problem.  I am only able to launch PrusaSlicer 2.8.1 as long as the Windows 10 VM has the 3D acceleration disabled.  Unfortunately, disabling 3D acceleration handicaps doing any 3D modeling tools create the models needed to feed PrusaSlicer.  This appears to be an issue with the 3D acceleration in VMware, but only PrusaSlicer seems to be triggering it.

timur-tabi, 2024-11-25:
Does this thread help? 

https://web.archive.org/web/20230402054809/https://communities.vmware.com/t5/VMware-Workstation-Pro/VMware-16-2-1-Display-3D-acceleration-not-working-Windows-10/m-p/2877574

pjvenda, 2025-10-21:
I am facing this problem exactly. VMWare host is on Windows 11, version is 17.6.4, guest OS is debian 13 and PrusaSlicer is 2.9.3.

3D acceleration disabled lets the software run but it is painfully slow and realistically unusable.

VMWare tools (open-vm-tools) are installed and up to date.

closed 2026-09-29 by prusaslicer-bot[bot]