prusa3d/PrusaSlicer #11736: [BUG] Prusa Slicer 2.7.0 (Latest) crashing while slicing full plate of objects with MK4 profile.

mixer3d, 2023-11-23:
### Description of the bug

While trying to slice the project using MK4 profile app is crashing. With MINI or XL profile i can fill the full plate and slice without issue. Object has 76 instances (for MK4), and 192 for XL. Spacing with autoalign option is set: 1mm, rotation allowed, geometry: accurate.    

[slicer-bug-report.txt](https://github.com/prusa3d/PrusaSlicer/files/13453879/slicer-bug-report.txt)

### Project file & How to reproduce

[slicer-bug-report.txt](https://github.com/prusa3d/PrusaSlicer/files/13453888/slicer-bug-report.txt)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.7.0+linux-x64-GTK3-202311231454.AppImage 

### Operating system

GNU/Debian 12.2

### Printer model

mk4

mixer3d, 2023-11-23:
the debug log only is attached

lukasmatena, 2023-11-24:
Please, attach a 3MF.

mixer3d, 2023-11-24:
Hi Lukas, i sorted out, the issue was related to that i had filament profile created for MINI+ in previous version on slicer, so alpha1 then alpha2 and beta, and this somehow stopped to work while checking between printing profiles, so after i recreated from scratch the filament profile (based on Generic PETG) i was able to slice the model failing before. 
So it was just related to some configuration mismatch, now after i removed custom materials and recreated them in the prod build, it works as expected. Thank you for support.

mixer3d, 2023-11-24:
I think it's resolved by itself after recreating the profile of one of custom materials which i made in alpha build before.

closed 2023-11-24 by mixer3d