prusa3d/PrusaSlicer #11737: Forgotten keyboard shortcut ?

ROBCON-Software, 2023-11-24:
Hi.

Keyboard shortcut for jump by 10 steps is forgotten to be mentioned in Help -> Keyboard shortcuts.

It applies to both horizontal and vertical sliders...

It is done by pressing Ctrl + Shift + arrow key (not only shift or ctrl)

Or change it to 5 on ctrl and 10 on shift.

Jan-Soustruznik, 2023-11-24:
Hello, @ROBCON-Software ,

move by 10 "steps"(is standard move without any shortcut, exactly move 10mm), there is another possible move with pressed keys " SHIFT + Any arrow" , that set move 1mm

All this info is already mentioned in  _Help -> Keyboard shortcuts -> Plater_

Why do you think that there should be useful add one more set move to 5mm?

Thank you, have a nice day

ROBCON-Software, 2023-11-27:
Hello Jane.

Perhaps You don't understand exactly...

In My slicer, there are these movements like default:

all arrow keys = move by 1 step
ctrl + any arrow = move by 5 steps
shift + anu arrow = move by 5 steps

and not mentioned in  Help -> Keyboard shortcuts:

ctrl + shift + any arrow = move by 10 steps

Also that in summary - there are 2 modifier keys with same step multiplier - ctrl and shif are same multipliers in slicer - also these produces movements multiplied to 5.

Also the idea is, why to have doubled multipliers, it would be better have one multiplier by 5 and another by 10, and not to have duplicate multipliers (ctrl, shif) and none for 10 (better say undocumented ctrl+shift).

It would be better have for example 5 by pressing ctrl multiplier key and 10 by pressing shift multiplier key.

Not to force users press both of these keys to become 10x multiplier...

Clear now ?

And still You can add this combined ctrl+shift key combo to other useful thing - like for example - move both arrows together also You can see a multiple layers cutout movable... not only one layer... You cannot drag any way both arrows (selection) at one time...

[image]

hkienle, 2024-01-27:
What I think would also be very useful is to have a shortcut to move the vertical slider all the way to the left -- or am I missing something?

closed 2026-09-29 by prusaslicer-bot[bot]