prusa3d/PrusaSlicer #11738: bgcode Upload via PrusaLink

the0bone, 2023-11-24:
### Description of the bug

Via the Windows Dialog by default only *.gcode files are visible.
No Filter for bgcode or both together. Only the way via *.* .

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 11

### Printer model

XL

JohnOCFII, 2023-11-25:
Might be similar to this issue I opened against the buddy firmware repository: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3502

rtyr, 2023-11-29:
This is a firmware issue and also already reported. Closing.

closed 2023-11-29 by rtyr