prusa3d/PrusaSlicer #11739: cut exist non manifold edge

Haidiye01, 2023-11-24:
### Description of the bug

Cutting the attachment directly will result in manifold edges

### Project file & How to reproduce

[HorizontalCut.zip](https://github.com/prusa3d/PrusaSlicer/files/13456538/HorizontalCut.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7

### Operating system

win10

### Printer model

voron 2.4

Haidiye01, 2023-11-24:
[image]

rtyr, 2023-11-24:
duplicate of your https://github.com/prusa3d/PrusaSlicer/issues/11740

closed 2023-11-24 by rtyr