prusa3d/PrusaSlicer #11741: "Reload from Disk - F5" did not work most of the time

yahbluez, 2023-11-24:
### Description of the bug

The menu function: 

"Edit -> Reload from Disk - F5"

Did still (version 2.7.0 all operating systems)  not work proper.
Most of the time nether the function key nor the menu entry works proper and as expected.

This very useful function for many workflows did most of the time nothing.

How to reproduce this bug:

- Start the slicer, 
- load a STL file from disk, 
- do a cut operation
- hit F5 or select the §reload from disk" entry

=> nothing happens.

Solution:

Let this function always reload the last loaded file(s) from.
If you think this may harm some users, just remove the F5 key and make this function at least from the menu do what it is supposed for to do.

### Project file & How to reproduce

How to reproduce this bug:

- Start the slicer, 
- load a STL file from disk, 
- do a cut operation
- hit F5 or select the §reload from disk" entry

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

all

### Printer model

all

yahbluez, 2023-11-25:
Hi @Jan-Soustruznik 

"Or do you think that the feature works different before?"

No i think that the function reload from disk should always reload from disk, 
no matter what was done with the model later.

It is all about user experience and workflow.

The way it is implemented now is so far away from being a useful workflow that i call that a bug.

A useful workflow is one that reduces the number of tasks a user has to repeat.
Here we have the situation that because user did something with the model (cut), the reload from disk did simply not work but the user has to do additional steps to archive what he likes to have, that the model is just reloaded as the name of the function suggested.

What is the idea, the sense behind the decision to switch off this feature because an additional step was done?

I know that UI is a kind of "not important" in prusaslicer development but this should change, UI is important.

I do not imagine any situation where it makes sense to not reload a model if the user likes to reload it.

Maybe that could be changed?

thx
yahbluez

Jan-Soustruznik, 2023-11-27:
Mainly this feature was implement for reload updated model out of Slicer not for recovery of modified model by slicer - for this purpose was implement undo/redo feature. When model mesh will be changed then "reload from disk" feature will be disabled (cut, scale, etc.. ), because the model doesn't mach with the original loaded model. Currently we do not plan to modify the "reload from disk" feature. 

I updated your request to Feature request.

yahbluez, 2023-11-27:
Ok, if one thinks about functionality and UI 
there is always the ctrl-Z that works as expected 
but a ctrl-z did not reload from disk.

I really do no see why a function should be named reload from disk if it not just did that.
In any other program i use reload from disk reloads the file no matter what else.

There is no use case where a "reload from disk, only if i have done nothing" makes any sense at all.

So taking this little bugfix as a feature request to make the prusaslicer better is a nice and good step.

lukasmatena, 2023-11-27:
> I really do no see why a function should be named reload from disk if it not just did that.

Cut function creates a new object. This object cannot be reloaded from disk, because it is not on the disk. The model would have to remember its original source and how it was cut (possibly repeatedly and with connectors) to make it work, and that is not implemented. It is not just a simple UI thing with no consequences.

> There is no use case where a "reload from disk, only if i have done nothing" makes any sense at all.

The main use case for reload from disk is for cases where you edit the stl in another software and want to reload the updated model. @Jan-Soustruznik has already told you that. And it actually makes a lot of sense.

yahbluez, 2023-11-27:
Hi @lukasmatena 

always interesting to see how many side effects a "small UI issue" may have.

How i would solve that, i see at least two ways:

If the object class has the attribute to store the path where it is loaded from,
let the child object inherits that attribute. (That could have side effects.)

Or a for sure side effect free way:
Store the last load / import path and let the 
reload from disk function just drop all objects 
and reload the last one. 
This is what is expected from a reload from disk function.

The workflow is indeed to change the STL external 
do for example a cut to see/measure details inside,
redo the object external 
and than reload what did not work now because of the cut.

There is no need to care about the new object born by cut,
it is obsolete in the moment the user needs to reload from disk
because the STL was already redone external.

closed 2026-09-29 by prusaslicer-bot[bot]