prusa3d/PrusaSlicer #11743: Dropdown menus getting brigther as I move the cursor

lucasfalcao3d, 2023-11-24:
### Description of the bug

As I move the cursor on the drop down menu, it gets brighter until the point it gets full white and I can't see a thing.

https://github.com/prusa3d/PrusaSlicer/assets/57116182/d2455bac-1f35-4296-8600-08d9e6e316dc

### Project file & How to reproduce

Check the attached video please.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.0+linux-x64-GTK3

### Operating system

Ubuntu 22.04.3 LTS x86_64

### Printer model

Prusa MINI+

lucasfalcao3d, 2023-11-24:
On the PrusaSlicer-2.7.0+linux-x64-**GTK2**-202311231518.AppImage this is not happening, although seem to be something wrong with the GTK theme, because in the 2.6.1 version of PrusaSlicer the menus looks like this.

[image]

Jan-Soustruznik, 2023-11-27:
Hello, @lucasfalcao3d , 

thank you for your report, could you share what custom Ubuntu theme do you using? Thank you

lucasfalcao3d, 2023-11-27:
Hey @Jan-Soustruznik ,

This one -> https://github.com/vinceliuice/Colloid-gtk-theme

Jan-Soustruznik, 2023-11-28:
SPE-2067

supvir452, 2023-11-29:
Would like to extend on the report above that the same is happening on my Arch KDE setup as well.
When I set GTK to the default the issue with the drop down list goes away.

Been running the same theme since PS 2.5.0 with no hassles ever until PS 2.7.0. 

I tested 2.5.0 and 2.6.0 on the default theme and both look the same when opening the dropdown lists but I notice that for 2.7.0 the drop down list looks different on the same default theme.

PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725:
[image]

PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220
[image]

PrusaSlicer-2.7.0+linux-x64-GTK3-202311231454
[image]

is it possible that the wxWidget component is a newer update perhaps ? i am taking wild shots in the dark here so please don't shoot me. :)

I have tried a few other apps running GTK3 and none gave this issue hence the thinking about the wxWidgets components being culprits. Alternatively I could get all the bits installed and run the source code and try and give clues as to what might be happening?

Really love PrusaSlicer and would not mind giving time to help and troubleshoot as a small thank you to all the work that goes into the  program.

---------

Version of PrusaSlicer
PrusaSlicer-2.7.0+linux-x64-GTK3-202311231454

Operating system
Arch Linux x64

muesli, 2023-12-16:
Can reproduce this with any theme other than the default GTK3 theme.

ArcaEge, 2023-12-17:
Same issue here as well - I am using the Kanagawa GTK theme on Fedora 39
[Screencast from 2023-12-17 17-00-31.webm](https://github.com/prusa3d/PrusaSlicer/assets/40526225/e7fd1075-427b-413e-925c-58f7b5a1549d)

wout, 2024-01-26:
Same issue here with the Appimage on elementary OS 7.1 (based on Ubuntu 22.04). Only, I'm using the light theme and the menu starts completely dark. I have to run over the menu several times with the cursor to make it legible. The snap and flatpak versions are fine.

Zinput, 2024-03-02:
@supvir452 I notice the same thing on my Arch/Sway setup. I was so annoyed that I tried many different packages of PrusaSlicer, only to find out that I'd just have to live with it to use a newer version of PrusaSlicer. Did you find any solutions?

Katherine1, 2024-04-13:
I can confirm this is happening with the Catppuccin theme for GTK as well.

JanisBerberi, 2024-05-06:
If you want to have workaround untill update is released, you can use alias for prusa-slicer:
In terminal type `alias prusa-slicer="GTK_THEME=elementary prusa-slicer"` or `alias prusa-slicer="GTK_THEME=Adwaita:dark prusa-slicer` for dark theme. It will launch with default theme and works as intended.

Zinput, 2024-10-20:
@JanisBerberi Thanks so much for the workaround! I went one step further and edited the Exec line of the desktop entry for PrusaSlicer to `env GTK_THEME=Adwaita:dark prusa-slicer %F`, and now the change also applies when launching the app outside the terminal.

closed 2026-09-29 by prusaslicer-bot[bot]