prusa3d/PrusaSlicer #11747: No virtual extruders in firmware 5.1.0

Blubbertje321, 2023-11-24:
### Description of the bug

In previous versions, you could use virtual extruders for multicolor printing. Since the 5.1.0 update for MK4, the printer get stuck when launching a gcode with more than 1 virtual extruder, displaying a black screen.

Make the printer compatible with multiple virtual extruders

### Project file & How to reproduce

[g code.txt](https://github.com/prusa3d/PrusaSlicer/files/13461798/g.code.txt)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 10

### Printer model

Prusa MK4

polarfar, 2023-11-26:
I have this problem too. Really annoying.

Blubbertje321, 2023-11-26:
> this is a prusaslicer problem or a firmware problem?

Firmware problem. Wrong section, oopss