prusa3d/PrusaSlicer #11748: Win arm64 - 2.7.0 x64 installer is no longer usable

docgalaxyblock, 2023-11-24:
### Description of the bug

Since the release of 2.7.0 the x64 installer exe from the prusa3d website errors on aarch64 windows on snapdragon devices.
On 2.6.1 it was working.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

prusa3d_win_2_7_0.exe

### Operating system

Windows 11 22H2 arm64

### Printer model

*not relevant*

lukasmatena, 2024-02-29:
@docgalaxyblock Is it working now with the 2.7.2 installer?

docgalaxyblock, 2024-02-29:
Yes it is working again 🎉

Thanks