prusa3d/PrusaSlicer #11751: Ability to associate print & filament presets with individual printers

3DPrintDemon, 2023-11-25:
**Is your feature request related to a problem? Please describe.**
When you have multiple different printers that use specific print settings & filament settings you have run through the list for all of them each time you slice anything between machines. Which is a real hassle. 

**Describe the solution you'd like**
I’d like to see the ability to associate print & filament presets to individual machines in the drop down menus

**Describe how it would work**
so that when you select a printer to slice for it’s designated presets are visible in the presets list at the top of the menu window while all other profiles can either be hidden or shown underneath these preferred & machine specific presets. This would save a lot time & improve workflow dramatically!

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

rtyr, 2023-11-25:
That's already possible. Use compatibility conditions.

3DPrintDemon, 2023-11-25:
> That's already possible. Use compatibility conditions.

Interesting, ok, In that case I’d like to request it becomes less of a hidden shadow feature, as I’ve never seen anything anywhere to suggest it’s possible, a little more foreground priority given would be lovely. Lol

rtyr, 2023-11-25:
It is not hidden. Look at the dependencies tab of any bundled printer profiles.

3DPrintDemon, 2023-11-25:
> That's already possible. Use compatibility conditions.

Interesting, will take a look for that one, obviously not seen that. Thanks for the tip. 
seems maybe it’s bit of a shadow feature.

rtyr, 2023-11-25:
One simple possible example:
Step 1) add text `SUPERPRINTER` to your printer notes (printer settings).

Step 2) add this compatible printer condition to your print and filament profiles meant for that printer:  `printer_notes=~/.*SUPERPRINTER.*/` (so those profiles will be visible only for printers, which have word SUPERPRINTER in the notes).

You can also combine conditions, for example:
`printer_notes=~/.*SUPERPRINTER.*/ and nozzle_diameter[0]==0.4`

closed 2023-11-25 by rtyr