prusa3d/PrusaSlicer #11754: Ramping lift can create travel moves out of bounds of the bed

rpls, 2023-11-25:
### Description of the bug

The new ramping travel generates a helical lift move, that may end up out of bounds if the starting point is near the edge of the bed.

### Project file & How to reproduce

[image]

[reproducer.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13465385/reproducer.3mf.zip)

Should be easy to reproduce, by placing an object very close to the edge. It's not much, but "120.017" is enough for my printer to cancel the print...

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0+MacOS-arm64

### Operating system

MacOS 14.1.1

### Printer model

Voron V0.1

Jan-Soustruznik, 2023-11-30:
Hello, @rpls , thank you for your report. We will try to fix in some next release. Have a nice day 

SPE-2068

lukasmatena, 2023-12-14:
Fixed in [PrusaSlicer 2.7.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1). Closing.

closed 2023-12-14 by lukasmatena