prusa3d/PrusaSlicer #11755: PrusaSlicer generates invalid gcode for Klipper with two-in-one-out set-up due to wrong use of SET_PRESSURE_ADVANCE gcode

NiklausHofer, 2023-11-26:
### Description of the bug

I have a printer with a two-in-one-ouet set up (two separates extruders feeding into a single, Y-shaped nozzle) running Klipper.

When slicing the popular [2 color tree frog](https://www.thingiverse.com/thing:329436) with PrusaSlicer and then priting, I am getting an error message, "Unable to infer active extruder" in Klipper which then aborts the print. The error occurs during the nozzle priming when then filament is changed for the first time.

This appears to be caused by the following line of gcode:
```gcode
 SET_PRESSURE_ADVANCE ADVANCE=0
```

According to [the Klipper gcode reference](https://www.klipper3d.org/G-Codes.html#set_pressure_advance) the `SET_PRESSURE_ADVANCE` takes an optional parameter for the extruder to be used in the action. I assume that due to the special set-up I have (one hotend with two extruders), Klipper can't figure out which stepper to drive w/o this parameter.

Unfortunately, the insertion of this gcode [is hardcoded in the slicer](https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/GCode/WipeTower.cpp#L106), so there is no way for me to fix this w/o recompiling.

(After manually patching the exported gcode, the print works just fine).

### Project file & How to reproduce

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/13465526/project.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Gentoo

### Printer model

custom (config in upload)

cdheiser, 2024-09-28:
+1.  This hardcoded option makes no sense.  Pressure advance is tuned in my printer.  I have to add extra code to the macros to reset pressure advance.

cdheiser, 2026-09-19:
I have checked the 3.0 alpha and it still has this problem.  There's no way to stop the disabling of pressure advance on the wipe tower.