prusa3d/PrusaSlicer #11756: Accessibility: Ability to see volumetric flow or speed (or other type) numeric value in preview by clicking with cursor on a line on model in preview

arekm, 2023-11-26:
**Is your feature request related to a problem? Please describe.**
When viewing preview, speed or volumetric flow rate Prusa Slicer nicely shows colours on the model for various values.
The problem is that close colours are not very distinguishable for people with some form of colour blindness or just with bad monitors.

**Describe the solution you'd like**
The solution would be ability to click on any line in the model and Prusa Slicer would show numerical value for currently selected type (like "volumetric flow rate").

Maybe just a line under all these values:

[image]

or appropriate value highlighted?

**Describe how it would work**
Click on model on yellow line and it would write or highlight "4.760".

**Describe alternatives you've considered**
Don't really see any alternative. Fixing eyes is not a viable option.

Tupson444, 2023-11-26:
I see pretty well and still can be difficult to distinguish what line is printed at what speed, because when there are some lines printed at in-between speeds so there isn't exact shade on the scale shown, you have to guess that a line is e.g. "something in-between 4.760 and 5.410", so you only end up with rough estimates instead of exact values.

neophyl, 2023-11-26:
This has been requested at least once already.  I remember seeing similar suggestions in other threads going back several years too.
For example https://github.com/prusa3d/PrusaSlicer/issues/7551  is the one that sticks in memory.

closed 2026-09-29 by prusaslicer-bot[bot]