prusa3d/PrusaSlicer #11759: Crash when selecting embossed text while SVG gizmo is opened

Noisyfox, 2023-11-26:
### Description of the bug

App crashes when selecting embossed text while SVG gizmo is opened, which it should not.

### Project file & How to reproduce

1. New project
2. Add an embossed text to the plate
3. Close the emboss gizmo
4. Right click on the empty area of the plate, then select add SVG
5. Add any svg, and keep the svg gizmo opened
6. Then trying selecting the embossed text
7. App crashes immediately

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0+win64

### Operating system

Windows 11

### Printer model

N/A

Jan-Soustruznik, 2023-11-27:
Hello, @Noisyfox ,

thank you for report. We try to fix in some next release. 

SPE-2058

lukasmatena, 2023-12-14:
Fixed in [PrusaSlicer 2.7.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1). Closing.