prusa3d/PrusaSlicer #11761: Helical travel on layer change before the retraction when retract on layer change is disabled

Bastel-Bodo, 2023-11-26:
### Description of the bug

PrusaSlicer 2.7.0 (release) outputs a "helical travel move" when changing layers, so that the "ramping travel lift" also applies here.
However, when "retract on layer change" is _disabled_ in the slicer settings, then the helical movement is still applied.
This causes the printer to extrude a little swirl on every layer change from the excess pressure in the nozzle.

This can even be observed when the next layer doesn't start where the previous one ended, and a retraction _is being_ output. In this case, the helical movement happens _before_ the retraction, so the printer still extrudes that little swirl.

### Project file & How to reproduce

You can observe the described behavior in the project files and GCodes in the attached ZIP archive.
In "X13T-Case-CameraFrame.3mf", the painted seams should allow to avoid travel moves entirely, yet a helical movement is still being rendered.
In "X13T-Case-Hex.3mf", a travel move is needed on layer change, causing a retract, but the retract happens _after_ the helical movement.

[Swirling extrusion.zip](https://github.com/prusa3d/PrusaSlicer/files/13466559/Swirling.extrusion.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0 release

### Operating system

Windows 10

### Printer model

Original Prusa MINI+

daniel-finger, 2023-11-26:
I can confirm the Issue, it took me a while to find out. This looked like this on my print: 
[image]
[image]

neophyl, 2023-11-26:
This looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11709 
Same was happening to me on my prints too.

Bastel-Bodo, 2023-11-26:
> This looks like a duplicate of #11709 Same was happening to me on my prints too.

Yes, kind of - though the linked issue is aiming for a different solution.
I my opinion, the helical move simply shouldn't happen when retract on layer change is disabled.
The linked issue, however, demands ramping travel to auto-enable retract on layer change - which isn't really a solution, because not retracting on layer change makes perfect sense for some prints where you may still want ramping travels for all the other travel moves within a layer.

lukasmatena, 2023-12-14:
Fixed in [PrusaSlicer 2.7.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1). Closing.

closed 2023-12-14 by lukasmatena

johngalt01, 2023-12-19:
please add a way to disable this in 2.7.2 or something i can not get it to turn off!

Bastel-Bodo, 2023-12-19:
@johngalt01 The helical travel move is tied to the "ramping travel lift" setting in printer settings (and filament override). If you disable that, helical travel is also disabled.