prusa3d/PrusaSlicer #11762: Fixed printing speed of the outer perimeter on all layers

MrSmokiMo, 2023-11-26:
Hello. The situation is as follows. The fact is that when printing, different printing speeds of the outer layer are used on different layers, which affects the quality of the outer layer by forming bulges or dips. But if the printing is going at the same speed, then there is no such problem. Is it possible to add the ability to print the outer walls at the same preset speed without taking into account acceleration and other modifiers?
[image]
[image]

neophyl, 2023-11-26:
What you mean like this ?
[image]

There are many settings already, if you tweak them its possible to do this already.

MrSmokiMo, 2023-11-27:
> What you mean like this ? [...]

Yes, dear friend, and how to achieve such a result?

Tupson444, 2023-11-27:
As far as I can see from the first image above, the speed variations are caused by minimum layer time settings (because of cooling): when there are only perimeters and infill, the layer time is too short so it slows down, but when there are top/bottom solid layers, the speed is less affected.

In that case, it would be good to have an option to exclude outer perimeters from the slowdown.

Jan-Soustruznik, 2023-11-27:
Hello, @MrSmokiMo, 

First, when you report any issue .. 1. Provide your 3MF file where is problem caused(we need it for debug); 2. what PS version using or version of PS where the problem is located; 3. OS platform ; 4. What printer do you use 

I think that you should to see on the **Prusa Knowledge Base**, there are some articles that may help you, example: 

Pressure equalizer: https://help.prusa3d.com/article/pressure-equalizer_331504#linear-advance-vs-pressure-equalizer
Speed settings: https://help.prusa3d.com/article/speed-settings_480325#dynamic-overhang-speed 
Cooling: https://help.prusa3d.com/article/cooling_127569

I closed issue, if you are tinking it is a program bug, create a new ticket. 

thank you, have a nice day