prusa3d/PrusaSlicer #11765: Prusaslicer Crashing

kapplion, 2023-11-26:
### Description of the bug

No matter which "complex" is model i try to slice prusaslicer crashes at 20% progress reliable. Simple geometry working fine, but at a certain size (scaling up) its crashing. I try it with PS 2.6, 2.6.1, 2.7 no differance.  I tryed differed printer and printer presets with diffent settings, same resoults. Seem like it crashes when a certain load gets to big and it crashes preemptively.

Cause I think it might be a thing with kind of a hardware issue here you have my PC components if this might be a thing:
CPU= 13th Gen Intel Core I9-139000KS
GPU= Gigabyte Nvidia Geforce RTX 4070 Ti
RAM= 2x 16GB Kingston DDR5-6400
Mainboard= Asus Z790 Gaming X AX

On my Laptop form work its just works fine (slow but fine ^^").

[Flexi Factory Skeleton T-Rex Body_Curved_Update.zip](https://github.com/prusa3d/PrusaSlicer/files/13467346/Flexi.Factory.Skeleton.T-Rex.Body_Curved_Update.zip)

[image]
[image]

### Project file & How to reproduce

Load the Project from the attachment and press "slice now"

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0+win64; Version 2.6.1+win64; Version 2.6.0+win64

### Operating system

Microsoft Windows 11 Pro

### Printer model

Prusa XL (Voron2.4, Voron V0 and others for testing)

w1ka, 2023-12-12:
Got the same problem

SckyzO, 2023-12-15:
I have the the same issue ...

Absolo13, 2023-12-18:
same issue with windows 11 and prusa 2.7.0

kapplion, 2023-12-20:
Prusaslicer 2.7.1 still same issue

kubispe1, 2023-12-22:
Hi, for my WIN 10 PS271. I am not able to reproduce it. 
We will look into it. Thanks

SckyzO, 2024-02-06:
> I am having the same issue with the latest 2.7.1, It as soon as I try to slice any stl, my computer restarts, never had this issue prior, system specs Intel 14th gen 19 14900k RTX 4090 128 GIG Dominator DDR5 Windows 11 64 bit.

Check here : https://github.com/prusa3d/PrusaSlicer/issues/11845

kapplion, 2024-02-29:
@SckyzO the thing with power limiting works great. I Capped at 200W. Your did it 300W. MIght this be a power supply issue? How big is your power supply? Mine is 650W.

SckyzO, 2024-02-29:
> @SckyzO the thing with power limiting works great. I Capped at 200W. Your did it 300W. MIght this be a power supply issue? How big is your power supply? Mine is 650W.

Yep ... Mine is 1000W. And work with 850w too.
Depend of your hardware 😁

closed 2024-03-04 by kapplion