prusa3d/PrusaSlicer #11770: GTK3 Dark theme: Combobox items are displayed dark on dark

trev0r-STA, 2023-11-27:
### Description of the bug

With the new version, the combobox item displayed black on dark gray background, it's not easy to read:

[image]

It's the same between the tar.bz2 or the AppImage. Combobox is OK in the GTK2 version (but not the about box!!!)
 
[image]

Seems there is some problem... :-)

My sytem: Debian 12 / MATE as Desktop Environment / theme: BlackMATE
Thanks for having a look

### Project file & How to reproduce

Just open the interface

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0+linux-x64-GTK3

### Operating system

Debian 12 (bookworm)

### Printer model

Not relevant in this case

Jan-Soustruznik, 2023-11-28:
SPE-2067

trev0r-STA, 2023-11-29:
All right. But... What does it mean ?

closed 2026-09-29 by prusaslicer-bot[bot]