prusa3d/PrusaSlicer #11772: Gcode generated for Z lift/hop does not respect Z speed limit

preyx, 2023-11-27:
### Description of the bug

I was attempting to print a complex part in my K1 with multiple z-hops, and I had a hotend crash while printing the first layer, destroying both the build plate and nozzle. When I sliced with Creality Print (Cura based), also with z-hops, I had no crash. Trying to figure out what's the difference between the two slicers as a cause for the inaccurate Z movement, I noticed this difference between the gcode generated for z-hop (gcode comments added manually for clarity):

PrusaSlicer:
```
G1 X93.099 Y88.012 E.04978     ; last extrusion of previous layer
G1 E-1 F2400                   ; retract
G1 X93.099 Y88.012 Z.34 F9000  ; z up
G1 X113.353 Y95.404            ; xy travel
G1 Z.24                        ; z down
G1 E1 F2400                    ; unretract
G1 F1800                       ; set layer feedrate
G1 X113.895 Y95.465 E.02246    ; first extrusion of next layer
```
Creality/Cura:
```
G1 X155.963 Y198.44 E3.01813         ; last extrusion of previous layer
G1 F2400 E3.01063                    ; retract
G1 F600 Z0.3                         ; z up
G0 F14400 X146.151 Y189.205 Z0.3     ; xy travel
G1 F600 Z0.2                         ; z down
G1 F2400 E3.51063                    ; unretract
G1 F3600 X147.623 Y187.733 E3.59613  ; first extrusion of next layer
```
The `F9000` correlates to a 150/mm speed, which is what I specified for X/Y travel speed, while my machine limit "Maximum Feedrate Z" is 10mm/s. Creality/Cura correctly generates this with `F600`. This issue appears in both my E3v2 config in PrusaSlicer, as well as my K1 config in the fork OrcaSlicer.

### Project file & How to reproduce

config file:
[e3v2config.zip](https://github.com/prusa3d/PrusaSlicer/files/13480706/e3v2config.zip)
3mf project file:
[test-Brick-Flexi-Conda_new_head.zip](https://github.com/prusa3d/PrusaSlicer/files/13480732/test-Brick-Flexi-Conda_new_head.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0+win64-202311231457_signed

### Operating system

Windows 10 22H2

### Printer model

Creality Ender 3v2 and K1

rtyr, 2023-11-28:
We are not limiting the feedrates in the gcode output, that should be handled by the firmware based either on it's default limits or by limits overriden by g-code command emitted by slicer (if "emit to g-code" option is enabled and if your FW supports it).

Anyway, you can set Z travel speed in print settings. As stated in the tooltip, slicer will use travel speed if Z speed is not specified.

[image]

preyx, 2023-11-28:
Thank you for the feedback. I missed that option when browsing settings. I'll review my firmware limits and set Z travel going forward for peace of mind. With a new search I found old PR https://github.com/prusa3d/PrusaSlicer/pull/6369 that added this feature and addressed this. Closing.