prusa3d/PrusaSlicer #11774: Multimaterial Painting produces incorrect slicing results

muesli, 2023-11-27:
### Description of the bug

Using "manual" multimaterial painting produces incorrect results for some objects after being sliced. See screenshots: the incorrect color is used for some perimeters.

Rotating the object just slightly fixes the slicing output.

[image]
[image]

### Project file & How to reproduce

[Extra_Printable_Companion_Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/13481280/Extra_Printable_Companion_Cube.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Ubuntu 22.04

### Printer model

Prusa XL

kubispe1, 2023-11-28:
We have it on priority list: SPE-1840

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena