prusa3d/PrusaSlicer #11775: PrusaSlicer 2.7 installer not working

gusdleon, 2023-11-27:
### Description of the bug

Starting the installer with a old install will result in a error on uninstall of the past install

### Project file & How to reproduce

Start the installer
[image]
Select yes in this prompt
[image]
working.....
[image]
The installer (uninstaller will fail)
[image]
with error code
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

windows 10 (19045.3693)

### Printer model

N/A (but mk3s)

kubispe1, 2023-11-28:
Thanks for your issue. We will try to fix and improve the installers.

Hovedskou, 2023-12-15:
I have the exact same issue.! Not able to install new versions after 2.6.1 due to this error.
Do we have any solutions yet..????

gusdleon, 2023-12-15:
> Do we have any solutions yet..????

You uninstall first PrusaSlicer 2.6.1 then install the new version

Hovedskou, 2023-12-15:
Already tried that. It give me the same error.! 😩venlig hilsenPreben Hovedskou
-------- Oprindelig besked --------Fra: Gustavo de León ***@***.***> Dato: 15.12.2023  13.47  (GMT+01:00) Til: prusa3d/PrusaSlicer ***@***.***> Cc: Hovedskou ***@***.***>, Comment ***@***.***> Emne: Re: [prusa3d/PrusaSlicer] PrusaSlicer 2.7 installer not working
  (Issue #11775) 

Do we have any solutions yet..????

You uninstall first PrusaSlicer 2.6.1 then install the new version

KtsCZE, 2024-02-16:
Same problem here. I did not find app Prusa in Settings > Apps. So I installed Prusa 2.6.1 again and after that uninstall. But problem still persist - I can´t still install version 2.7.1 with same errors

closed 2026-09-29 by prusaslicer-bot[bot]