prusa3d/PrusaSlicer #11777: Cannot prime all extruders unless you use a Wipe tower

BrienKing, 2023-11-28:
**Is your feature request related to a problem? Please describe.**
Yes, with some multi-material prints I don't use a wipe tower because it's not needed.   However, I would like all extruders to be primed at the start of a print.   I had one failed print because the second extruder wasn't extruding at the start of a print.

**Describe the solution you'd like**
Make "Prime All Printing Extruders" option available without the need for a Prime Tower.

**Describe how it would work**
Should be self explanatory. 

**Describe alternatives you've considered**
Hand coding G-Code to do it.

kubispe1, 2024-04-03:
When I use an XL printer with the prime tower disabled, all the extruders that are used get primed at the very beginning of the print.
This is likely due to custom G-code, even though it seems to function correctly.
Is this what you were referring to?
I apologize, it's been a while since I looked into this issue, so I might have missed the key details.

metaxis, 2025-05-21:
Yes, I can confirm that there is a legitimate reason to want to prime all in-use toolheads but not have a wipe tower.  My 3rd party IDEX revos do not ooze while parked, so I don't need a wipe/prime tower and thus don't use the slicer's built-in functionality.    

I ended up "borrowing" and adopting the XL's extensive start gcode to smart-prime the extruders, which is fine but feels a bit like an anti-pattern, especially with redundant funtionality right there.   Using the slicer would greatly simplify the printer config.

closed 2026-09-29 by prusaslicer-bot[bot]