prusa3d/PrusaSlicer #11785: Arrange options not visible for the XL with ver 2.7.0

DoubleStrike, 2023-11-29:
### Description of the bug

I am using the latest 2.7.0 release version.  According to the release notes blog post, I should be able to right-click on the arrange toolbar icon and get options for which corner to align to on the XL.

However that option (which used to appear in the alpha and beta version) is no longer there.  I only have a single printer profile installed for the XL5 with input shaping.

### Project file & How to reproduce

Right click on Arrange icon.  I should see a preferred corner or center choice, but it is missing.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0+win64

### Operating system

Windows 10 22H2

### Printer model

XL 5-toolhead with input shaper

LuckyResistor, 2023-12-17:
Same here, with the latest version:
Version 2.7.1+MacOS-arm64

[image]

Jan-Soustruznik, 2024-01-03:
Hello, @DoubleStrike , 

Additional Alignment option is available only for one-tool XL printer. For multi-tool XL is set default "Rear center".

DoubleStrike, 2024-01-04:
> Hello, @DoubleStrike ,
> 
> Additional Alignment option is available only for one-tool XL printer. For multi-tool XL is set default "Rear center".

Weirdly enough, it's not apparently that simple.  I have been doing a lot of testing, and sometimes that option appears for me on my 5 toolhead printer (using the 5 toolhead profile) and sometimes it does not.  I can't find any rhyme or reason for when it does appear vs not. Most of my prints are single color, single tool prints, but it's not consistent.

DoubleStrike, 2024-01-04:
Here's the picture 10 seconds later where it disappeared again:
[image]

stefanboca, 2024-04-04:
I've been having the same issue on PrusaSlicer 2.7.3 with the 5 toolhead XL profile.

ex-nerd, 2024-07-20:
Seems to be a duplicate of #12515 and my original #11414

And for what it's worth, this is still happening with 2.8.

Why should the behavior be different for single- vs multi-tool machines? I thought the entire point of arranging parts in one corner was to reduce power consumption from the heated bed tiles. Even if this is about optimizing prints for multiple materials, that doesn't mean that I don't predominantly use my 5-tool XL for single-color prints, so the behavior should depend on what I'm printing, not what my printer is physically capable of. Even then, printing in a rear corner has more benefits for energy savings.

Regardless of any of this, there is no reason the menu option should disappear and reappear based on whether or not I've clicked "slice" once and then not changed any settings. Only the default should change, but that should not prevent me from overriding that default just like I can do for spacing, etc.

ex-nerd, 2024-08-30:
This seems to be tied to whether or not there is a wipe/purge tower. E.g:

The tower exists on my 5-tool XL even if I'm just doing a single-material print. When the tower exists, the alignment menu option is not present and everything automatically arranges at the rear center of the print area:

[image]

However, once I slice the model and the system determines the tower is unnecessary, it's removed. When I return to the layout screen, the alignment menu option returns and I can select "front left" as has been recommended for more efficient power usage:

[image]

Annoyingly, there aren't options to prefer front/rear alignment, which could save a bit of power consumption even with the tower in there (the existing alignment option seems to be "radially from the rear center point" rather than "along the rear edge" that would keep parts compact and reduce the number of "nearby" tiles that need to be heated up).

ex-nerd, 2026-09-18:
This still seems to be the case with v3 alpha. The `Alignment` option disappears when the purge tower exists:

[image]
[image]

v3 makes it even worse because as far as I can see, the option to disable the purge tower is no longer accessible under the multi-extruder advanced settings section.

Jan-Soustruznik, 2026-10-21:
Hello, This is actually expected behavior. The alignment feature is only available without a wipe tower.

closed 2026-10-21 by Jan-Soustruznik

ex-nerd, 2026-10-21:
> 15\. alignment feature is only available without a wipe tower

Really?! Why would you do that? In that case, please mark this as a feature request. It's terrible/confusing behavior to randomly hide part of the UX based on some unrelated config option.

It's such a hassle having to disable the tower, align my parts to whatever section of the bed best fits the print (or simply whichever corner I want in order to reduce wear on the print sheet), re-enable the tower and then manually move it into a good spot for the alignment. Rear center isn't always the most ideal spot.